vilemduha/blendercam #87: CAM Material

bostjannagode, 2016-09-13:
Hello, 
first thanks again for great work. Just a little question. Is it possible to use one of objects as source in CAM material Size and Position?? I mean, can it be iregullar shape in any of future versions? Would be great if possible..
Best, Bostjan Nagode

jeffmd, 2016-09-14:
That's a good idea.  Can't see why it can't be done.

bostjannagode, 2016-09-14:
BIG THANKS. SOON?

Best, Bostjan Nagode

bostjannagode, 2016-09-14:
Why? As you know, you do rough first with big toolpath distance. Some leftovers can be pretty big. So, you will damage on them. If you could say material will be simulation of previous rough cut, wou yould not brake anything..
Best, Bostjan

Will post picture of milling ALU casting when will be done (few days)...

ksuprynowicz, 2016-09-18:
Such feature would also save a lot of time on machining large workpieces with a small details.
I think that for drop cutter operations the easiest way to do it would be to sample both material model and source model and then remove parts of toolpath that are not inside material model.

closed 2023-12-14 by pppalain