vilemduha/blendercam #90: Is BlenderCAM going to be compatible with the latests versions of Blender, anytime soon?

Petross404, 2016-10-02:
Good afternoon. I like very much BlenderCAM and wrote an ebuild for it and I thought to share it with the Gentoo community. But I noticed that this addon only works with blender-2.76 exactly, but soon versions 2.77 and 2.78 are going to be ported into the Portage tree. So I want to know what is the future of blenderCAM in order to not force an (rather) outdated blender version.

Keep up the good work :-)

ksuprynowicz, 2016-10-03:
Hi!
I'm using github version of BlenderCAM with Blender 2.77a(packaged in Debian Testing) and it's definitely compatible. I haven't checked yet with 2.78(there isn't one in Debian yet), but I don't expect too many problems.

ksuprynowicz, 2016-10-03:
Also the current OpenCAMLib support in BlenderCAM works only with 2.77a(or probably with newer versions), so I don't recommend 2.76.

Petross404, 2016-10-03:
Thanks for the response :)

Now, I tried with Blender-2.78/2.76 ("build" on gentoo to use python3-5) and when I tried to enable the addon, I got an error about a MultiArray that couldn't be imported. When I tried blender-2.76 and python3.4 it enabled just ok. 

Should it be an issue with Shapely that is also build to use python3.4 and not specific to blender?

ksuprynowicz, 2016-10-03:
When you built Blender 2.78 with python 3.5, did you also built Shapely with python 3.5 or did you use 3.4 build?

Petross404, 2016-10-03:
I built Shapely with python3.4. I am afraid there isn't a python3.5 option (USE flag) for Shapely in the portage. Will open a bug for it :-)

Do you think that is the reason for cannot import MultiArray?

ksuprynowicz, 2016-10-03:
I think that it's the most probable reason. As soon as 2.78 appears in Debian Testing, I'll give it a try.
Python 3.5 definitely isn't a problem, because in Debian Blender 2.77a uses python 3.5, and everything works perfectly.

jeffmd, 2016-10-03:
I am using blendercam with blender built from git master each day + python 3.5 with no problems on linux.

benjaminjames, 2018-09-10:
I'm finding that the issues between which version of Blender and which version of python to use make this package infeasible. I've tried a few combinations and still get a "no module named Shapely". 4 hours of wrestling with it..

benjaminjames, 2018-10-10:
> If you are using Linux, I suggest using Blender and Shapely version from your package manager. Normally Blender distribution has it's own Python installation. When you use Blender from package manager, it uses system-wide Python, so you just install Shapely package..

Sorry I'm on a mac and I really have no idea what any of that means.

pppalain, 2023-01-14:
blendercam is compatible with blender 3.14 as of writing

closed 2023-01-14 by pppalain