vilemduha/blendercam #91: tag 0.9 release please

greyltc, 2016-11-01:
Hi,
It looks like 0.9 is out:
http://blendercam.blogspot.co.uk/

Could you please tag the release appropriately here so that I know what code is stable?

Thanks!

vilemnovak, 2017-02-20:
sorry, we never tagged, we rather make downloadable packages... I actually don't know how to tag :) Maybe I find out, but it should be with next versin.

greyltc, 2017-02-21:
It's super easy!
1. Choose a commit that you'd like to mark as stable and note down its commit hash
2. Visit https://github.com/vilemnovak/blendercam/releases
3. Click the `Target:master` button and paste in your commit hash from step 1
4. Specify a version
5. Optionally fill in the rest of the info
6. Click the `Publish release` button

pppalain, 2023-01-14:
still not tagged

closed 2023-01-14 by pppalain