vilemduha/blendercam #94: please help i dont know what to do

nareknc, 2016-12-22:
[image]

narvf, 2016-12-22:
Can you please explain, what problem do you have? What is your operating system? How did you install/download blendercam?

nareknc, 2016-12-22:
operating system is ubuntu 14.04.1 and i download and install blendercam  from here https://github.com/vilemnovak/blendercam/wiki/Using-Blendercam-from-github#enable-blendercam-add-on 
and now i dont understand what i am doing wrong

narvf, 2016-12-22:
Try downloading binary from site blendercam.blogspot.com, https://drive.google.com/drive/folders/0B8_m8tGvVyKpYUVTMm14YWxTWEk
Make sure you have installed:
python-shapely
python-numpy
I don't clearly see the problem. You can not enable plugins? Does blender show some errors?

nareknc, 2016-12-23:
here is the error 
[image]

narvf, 2016-12-23:
Look slike you don't have installed python-shapely
Maybe ubuntu 14.10 doesn't have this package?

vilemnovak, 2017-02-20:
Hi, Indeed on ubuntu you need to go to synaptic package manager and install shapely. Since this isn't a bug, closing issue ;)

closed 2017-02-20 by vilemnovak