vilemduha/blendercam #95: Finish toolpath

fazekas1974, 2017-01-08:
Hello!
I wanted to make a tutorial finishing tool path, according to the Rough tool path.
Made it but did not look around the object, and completed G-code is only four lines. I attach pictures of the options.
Where can I make a mistake?
[image]
[image]
[image]
[image]
[image]
[image]
[Vinerfinish.txt](https://github.com/vilemnovak/blendercam/files/692067/Vinerfinish.txt)

narvf, 2017-01-08:
I don't think that i can help - since I mostly used bcam for 2d toolpath generation, but please post blend file - maybe someone else will look into.

fazekas1974, 2017-01-08:
The post my pictures there amid the Vinerfinish.txt.
The rough toolpath generated finish tool path.

vilemnovak, 2017-02-20:
Hi, the problem is that you have a cam path as in input object, put there your source mesh object.
Closing this because it's not a bug.

closed 2017-02-20 by vilemnovak