vilemduha/blendercam #97: spherical cavity

kylotron, 2017-02-17:
Hello to all.

I realize the spherical cavity to stay of 6.2 mm diameter steel balls. How should I do?

[image]

[probe1b.blend.zip](https://github.com/vilemnovak/blendercam/files/783236/probe1b.blend.zip)

narvf, 2017-02-18:
Maybe try pocket operation with ball nose? But I am not sure if this will suits your needs.

vilemnovak, 2017-02-20:
If this should be precise, and mass produced, best would be to get a spherical cutter of that size?
otherwise, try any of the 3d operations, but add more detail to the spherical surfaces.

vilemnovak, 2017-02-20:
also closing, not a bug or issue, we have forum.

closed 2017-02-20 by vilemnovak