vilemduha/blendercam #99: not generating /tmp/ocl_chunk_samples.txt

TurBoss, 2017-04-14:
Hi:
I hit this error
```
progress{building path pattern}
progress{22.006794929504395}
Traceback (most recent call last):
  File "/home/turboss/PortableApps/blender/blendercam/scripts/addons/cam/opencamlib/oclSample.py", line 7, in <module>
    stl = camvtk.STLSurf(os.path.join(tempfile.gettempdir(), "model0.stl"))
  File "/usr/lib/python2.7/dist-packages/camvtk.py", line 725, in __init__
    self.mapper.SetInput(self.src.GetOutput())
AttributeError: SetInput
Traceback (most recent call last):
  File "/home/turboss/PortableApps/blender/blendercam/scripts/addons/cam/ops.py", line 219, in execute
    utils.getPath(context,o)
  File "/home/turboss/PortableApps/blender/blendercam/scripts/addons/cam/utils.py", line 3356, in getPath
    getPath3axis(context,operation)
  File "/home/turboss/PortableApps/blender/blendercam/scripts/addons/cam/utils.py", line 2922, in getPath3axis
    chunks.extend(sampleChunks(o,pathSamples,layers))
  File "/home/turboss/PortableApps/blender/blendercam/scripts/addons/cam/utils.py", line 340, in sampleChunks
    oclSample(o, pathSamples)
  File "/home/turboss/PortableApps/blender/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 133, in oclSample
    chunkPointSamplesFromCSV(chunks)
  File "/home/turboss/PortableApps/blender/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 60, in chunkPointSamplesFromCSV
    csv_file = open(os.path.join(tempfile.gettempdir(), 'ocl_chunk_samples.txt'), 'r')
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/ocl_chunk_samples.txt'

location: <unknown location>:-1

location: <unknown location>:-1
```

I can't figure out how to fix

thanks

ksuprynowicz, 2017-04-17:
Hi!
What version of vtk do you have?

TurBoss, 2017-04-17:
How I can see it? I'm on ubuntu 16.10 is VTK6?

i have ocl for the 5 will try to compile the fork for the 6 

sorry my bad

ksuprynowicz, 2017-04-17:
I think that may be the problem :)

TurBoss, 2017-04-17:
does $ make package work?

ksuprynowicz, 2017-04-17:
I haven't tried, but if it doesn't you can modify original version to work with VTK6 just by swapping camvtk.py from modified one in my repository:

https://github.com/ksuprynowicz/opencamlib

TurBoss, 2017-04-17:
ok deb installed succesfully now im goin to try the real deal

ksuprynowicz, 2017-04-17:
I compile and install it by 'make' and 'make install' command. Building a package would definitely be a better way, but I don't have experience with it.

ksuprynowicz, 2017-04-17:
Let me know if it works :)

TurBoss, 2017-04-17:
sure

TurBoss, 2017-04-17:
this is what I get : 
```
$ ./blender
Read new prefs: /home/turboss/.config/blender/2.78/config/userpref.blend
found bundled python: /home/turboss/PortableApps/blender/blender-2.78c-linux-glibc219-x86_64/2.78/python
progress{building path pattern}
progress{16.949191331863403}
Traceback (most recent call last):
  File "/home/turboss/PortableApps/blender/blendercam/scripts/addons/cam/opencamlib/oclSample.py", line 2, in <module>
    import ocl
ImportError: /usr/lib/python2.7/dist-packages/ocl.so: undefined symbol: _ZN5boost6python6detail11init_moduleER11PyModuleDefPFvvE
Traceback (most recent call last):
  File "/home/turboss/PortableApps/blender/blendercam/scripts/addons/cam/ops.py", line 219, in execute
    utils.getPath(context,o)
  File "/home/turboss/PortableApps/blender/blendercam/scripts/addons/cam/utils.py", line 3356, in getPath
    getPath3axis(context,operation)
  File "/home/turboss/PortableApps/blender/blendercam/scripts/addons/cam/utils.py", line 2922, in getPath3axis
    chunks.extend(sampleChunks(o,pathSamples,layers))
  File "/home/turboss/PortableApps/blender/blendercam/scripts/addons/cam/utils.py", line 340, in sampleChunks
    oclSample(o, pathSamples)
  File "/home/turboss/PortableApps/blender/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 133, in oclSample
    chunkPointSamplesFromCSV(chunks)
  File "/home/turboss/PortableApps/blender/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 60, in chunkPointSamplesFromCSV
    csv_file = open(os.path.join(tempfile.gettempdir(), 'ocl_chunk_samples.txt'), 'r')
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/ocl_chunk_samples.txt'

location: <unknown location>:-1

location: <unknown location>:-1
```

I'm goin to try with make install

TurBoss, 2017-04-17:
nope the same error

ksuprynowicz, 2017-04-17:
It's a bit different now, last time error was in lines:

    self.mapper.SetInput(self.src.GetOutput())
    AttributeError: SetInput

And now it's:

    ImportError: /usr/lib/python2.7/dist-packages/ocl.so: undefined symbol: _ZN5boost6python6detail11init_moduleER11PyModuleDefPFvvE

Do you have more than one installation of boost/boost python in library path(including /usr/local/lib and other directories)?

TurBoss, 2017-04-17:
aha i catch it cmake uses python 3.5 
this was fixed in mainstream
I'm recompiling with 2.7 set in cmake

TurBoss, 2017-04-17:
Thats it

Thanks 

```turboss@turbo-box:~/PortableApps/blender/blender-2.78c-linux-glibc219-x86_64$ ./blender
Read new prefs: /home/turboss/.config/blender/2.78/config/userpref.blend
found bundled python: /home/turboss/PortableApps/blender/blender-2.78c-linux-glibc219-x86_64/2.78/python
read blend: /home/turboss/Projects/Tiles/cyber_00.blend
progress{building path pattern}
progress{26.557661294937134}
bdc::setSTL()
bdc::setSTL() done.
dropCutterSTL5 1414747 cl-points and 524288 triangles.

0%   10   20   30   40   50   60   70   80   90   100%
|----|----|----|----|----|----|----|----|----|----|
Number of OpenMP threads = 2
*
```

closed 2017-04-17 by TurBoss

TurBoss, 2017-04-18:
forgot to mention that is working with make package