vilemduha/blendercam #100: Show progress as percentage bar in blender ui. Add button to stop computations.

Harvie, 2017-05-03:
Hello,
sometimes i forgot to open blender from commandline and therefore i don't have any way to tell how long it will take to finish path calculation. Also there's no way to stop it once i click "calculate path". Can you do this to work same way as blender rendering usualy works? There's progressbar on top of blender window and button to stop the progress if it takes too long or something... The way it's now can be quite annoying...

zohozer, 2017-05-04:
Yes, indeed, this will be a nice feature I would also love to see.

abosafia, 2024-01-28:
There is a new way to install blendercam
https://github.com/pppalain/blendercam/releases/tag/0.9.11
Follow instructions.
In that release there is a progress in the bottom left corner. And you can press esc to terminate the operation.

closed 2024-04-11 by pppalain