vilemduha/blendercam #101: Compatibility with Blender Net Render?

Harvie, 2017-05-03:
I think it would be cool if BlenderCAM was able to process slow operations using blender netrender infrastructure on remote blender servers... I use this to speed up slow blender renders. And even PyCAM is able to run cluster of PyCAM servers which are then queried from client's GUI.

ksuprynowicz, 2017-08-08:
I think it could be quite easy to do - I don't know much about Blender Net Render, but currently in BlenderCAM, when you use 'calculate in background' function, another instance of Blender is started to compute operation trajectory. The easiest way would be to add option for starting another instance not locally, but on remote server, using ssh. To copy files and results you could use scp, or just share folder with sshfs.

Harvie, 2017-08-09:
@ksuprynowicz there already is infrastructure for this in blender, so no need to go low-level (eg. with ssh/sshfs). However i think blender net render also supports clustering of some kind, so it might be vital if it was possible to split calculation in several independent tasks, so it can be run on multiple servers at once. I am not aware how well is blendercam suited for parallel computations (even locally on multiple CPU cores) right now.

However some initial support for netrender would be ok per se. It basically just automaticaly transfers files on remote machine and starts bake/render like one would do manualy.

ksuprynowicz, 2017-08-09:
Operations in BlenderCAM are computed in several steps, the most computation-intensive one is sampling cutter-model contact points. It can be parallelized or offloaded to multiple machines quite easily. Right now if you want to compute operation on multiple cores, you just have to enable OpenCAMLib support, as described here:

https://github.com/vilemduha/blendercam/wiki/Using-BlenderCAM-with-OpenCAMLib

Additionally from enabling OpenCAMLib support you will get higher quality toolpaths, especially waterlines. I've tested it on 24 physical core Opteron machine, and for some cases computations are up to 40 times faster. If you want to offload it to another machine (and I think it's a great idea), take a look at the sources in:

https://github.com/vilemduha/blendercam/tree/master/scripts/addons/cam/opencamlib

Steps other than sampling would be very difficult to paralellize or offload to multiple machines(offloading to single machine should be possible). For local parallelization on multicore CPUs the most important problem is Python's Global Interpreter Lock. Another problem is that these steps(like curve offsetting) don't divide as easily into smaller separate pieces.

If you have any problems with OpenCAMLib support in BlenderCAM, feel free to ask :)

Harvie, 2017-08-09:
> the most computation-intensive one is sampling cutter-model contact points. It can be parallelized or offloaded to multiple machines quite easily.

@ksuprynowicz that's great news!

closed 2023-01-14 by pppalain