vilemduha/blendercam #105: opencamlib exception

TimPut, 2017-06-09:
On a clean arch install with Blender 2.78, and installing opencamlib-git and blendercam-git from the AUR, opencamlib raises this exception when used: 
``
Traceback (most recent call last):
  File "/usr/share/blender/2.78/scripts/addons/cam/ops.py", line 219, in execute
    utils.getPath(context,o)
  File "/usr/share/blender/2.78/scripts/addons/cam/utils.py", line 3356, in getPath
    getPath3axis(context,operation)
  File "/usr/share/blender/2.78/scripts/addons/cam/utils.py", line 2922, in getPath3axis
    chunks.extend(sampleChunks(o,pathSamples,layers))
  File "/usr/share/blender/2.78/scripts/addons/cam/utils.py", line 340, in sampleChunks
    oclSample(o, pathSamples)
  File "/usr/share/blender/2.78/scripts/addons/cam/opencamlib/opencamlib.py", line 114, in oclSample
    if os.path.isdir(os.path.join(bpy.utils.script_path_pref(), "addons", "cam", "opencamlib")):
  File "/usr/lib/python3.6/posixpath.py", line 78, in join
    a = os.fspath(a)
TypeError: expected str, bytes or os.PathLike object, not NoneType
``

TurBoss, 2017-06-09:
can you try with this vtk6 compatible opencamlib?
https://github.com/ksuprynowicz/opencamlib

TurBoss, 2017-06-09:
also specify in cmake python 2.7

Harvie, 2019-03-12:
I have the same problem:

```
/usr/share/blender/2.79/scripts/addons/cam/ops.py:71
rna_uiItemPointerR: property not found: OBJECT_OT_cam_position.operation
/usr/share/blender/2.79/scripts/addons/cam/ops.py:71
changing sampling resolution to 0.000619
operation: cutout
object method for retrieving silhouette
shapely getting silhouette
0.20934391021728516
Polygon 1
[<shapely.geometry.polygon.Polygon object at 0x7f036a349a58>]
progress{sorting paths}
Traceback (most recent call last):
  File "/usr/share/blender/2.79/scripts/addons/cam/ops.py", line 219, in execute
    utils.getPath(context,o)
  File "/usr/share/blender/2.79/scripts/addons/cam/utils.py", line 3356, in getPath
    getPath3axis(context,operation)
  File "/usr/share/blender/2.79/scripts/addons/cam/utils.py", line 2862, in getPath3axis
    strategy_cutout( o )
  File "/usr/share/blender/2.79/scripts/addons/cam/utils.py", line 2253, in strategy_cutout
    chunksFromCurve=sortChunks(chunksFromCurve,o)
  File "/usr/share/blender/2.79/scripts/addons/cam/utils.py", line 1594, in sortChunks
    sortedchunks = connectChunksLow(sortedchunks,o)
  File "/usr/share/blender/2.79/scripts/addons/cam/utils.py", line 1509, in connectChunksLow
    oclResampleChunks(o, chunks_to_resample)
  File "/usr/share/blender/2.79/scripts/addons/cam/opencamlib/opencamlib.py", line 124, in oclResampleChunks
    if os.path.isdir(os.path.join(bpy.utils.script_path_pref(), "addons", "cam", "opencamlib")):
  File "/usr/lib/python3.7/posixpath.py", line 80, in join
    a = os.fspath(a)
TypeError: expected str, bytes or os.PathLike object, not NoneType

location: <unknown location>:-1

location: <unknown location>:-1
rna_uiItemPointerR: property not found: OBJECT_OT_cam_position.operation
/usr/share/blender/2.79/scripts/addons/cam/ops.py:71

```

pppalain, 2023-01-14:
old old issue

closed 2023-01-14 by pppalain