vilemduha/blendercam #106: error when using opencamlib option

magveda, 2017-08-22:
could you please help me. I get the error:

```
progress{building path pattern}
progress{0.6838352680206299}
Traceback (most recent call last):
  File "/home/nemo/bin/blendercam/scripts/addons/cam/opencamlib/oclSample.py", line 4, in <module>
    import pyocl
ImportError: No module named pyocl
Traceback (most recent call last):
  File "/home/nemo/bin/blendercam/scripts/addons/cam/ops.py", line 219, in execute
    utils.getPath(context,o)
  File "/home/nemo/bin/blendercam/scripts/addons/cam/utils.py", line 3356, in getPath
    getPath3axis(context,operation)
  File "/home/nemo/bin/blendercam/scripts/addons/cam/utils.py", line 2922, in getPath3axis
    chunks.extend(sampleChunks(o,pathSamples,layers))
  File "/home/nemo/bin/blendercam/scripts/addons/cam/utils.py", line 340, in sampleChunks
    oclSample(o, pathSamples)
  File "/home/nemo/bin/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 118, in oclSample
    chunkPointSamplesFromCSV(chunks)
  File "/home/nemo/bin/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 55, in chunkPointSamplesFromCSV
    csv_file = open(os.path.join(tempfile.gettempdir(), 'ocl_chunk_samples.txt'),'r')
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/ocl_chunk_samples.txt'

location: <unknown location>:-1

location: <unknown location>:-1

```

when I issue these commands on terminal I get:

```
python
Python 2.7.5 (default, Nov  6 2016, 00:28:07) 
[GCC 4.8.5 20150623 (Red Hat 4.8.5-11)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import ocl
>>> import pyocl
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named pyocl
>>>
```

where should I get `pyocl`?

magveda, 2017-08-22:
after some compiling options error changed to:

```
progress{building path pattern}
progress{0.8857207298278809}
Traceback (most recent call last):
  File "/home/nemo/bin/blendercam/scripts/addons/cam/opencamlib/oclSample.py", line 7, in <module>
    stl = camvtk.STLSurf(os.path.join(tempfile.gettempdir(), "model0.stl"))
  File "/usr/lib/python2.7/site-packages/camvtk.py", line 725, in __init__
    self.mapper.SetInput(self.src.GetOutput())
AttributeError: SetInput
Traceback (most recent call last):
  File "/home/nemo/bin/blendercam/scripts/addons/cam/ops.py", line 219, in execute
    utils.getPath(context,o)
  File "/home/nemo/bin/blendercam/scripts/addons/cam/utils.py", line 3356, in getPath
    getPath3axis(context,operation)
  File "/home/nemo/bin/blendercam/scripts/addons/cam/utils.py", line 2922, in getPath3axis
    chunks.extend(sampleChunks(o,pathSamples,layers))
  File "/home/nemo/bin/blendercam/scripts/addons/cam/utils.py", line 340, in sampleChunks
    oclSample(o, pathSamples)
  File "/home/nemo/bin/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 118, in oclSample
    chunkPointSamplesFromCSV(chunks)
  File "/home/nemo/bin/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 55, in chunkPointSamplesFromCSV
    csv_file = open(os.path.join(tempfile.gettempdir(), 'ocl_chunk_samples.txt'),'r')
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/ocl_chunk_samples.txt'

location: <unknown location>:-1

location: <unknown location>:-1
```

magveda, 2017-08-22:
second error was with VTK6

```
sudo vim /usr/lib/python2.7/site-packages/camvtk.py
```
replaced line 725 from
```
self.mapper.SetInput(self.src.GetOutput())
```
to
```
self.mapper.SetInputData(self.src.GetOutput())
```

more info: http://www.vtk.org/Wiki/VTK/VTK_6_Migration/Replacement_of_SetInput

closed 2017-08-22 by magveda

ksuprynowicz, 2017-08-23:
VTK6 version of OpenCAMLib can be found here:

https://github.com/ksuprynowicz/opencamlib

magveda, 2017-08-24:
I cloned and compiled the latest version from https://github.com/aewallin/opencamlib.git

and adapted it by modifying one line of code.

when I tried to do from https://github.com/ksuprynowicz/opencamlib I got many errors about some bad strings...