vilemduha/blendercam #107: paths for vertical walls are not vertical

magveda, 2017-08-23:
hey, thank you for this great tool!

I want to mill a small PCB holder and the stand pins are not milled vertical - how can I improve? also they are not round.. what are the steps to make gcode first go round about all the pins?..

[image]

[image]

ksuprynowicz, 2017-08-23:
Hi!
The fastest way would be using OpenCAMLib waterlines. Here you can find info on how to install it:

 https://github.com/vilemduha/blendercam/wiki/Using-BlenderCAM-with-OpenCAMLib

For shapes like the one in your post, you will get best results by using and applying 'edge split' modifier and using 'select non manifold' option in edit mode to get 2d curves. Then you can use Pocket and Profile operations for roughing and finishing cut.

If you have any questions, feel free to ask :)

ksuprynowicz, 2017-08-23:
And as for 'Parallel' and other vertically sampled operations, you can improve results by decreasing sampling distance along paths. 0,02mm seems like good value for most stuff.

magveda, 2017-08-24:
Thank you for you answer!

I have OpenCAMLib and it is working with `Parallel`. When I change to `Waterlines` I get

```
Traceback (most recent call last):
  File "/home/nemo/bin/blendercam/scripts/addons/cam/ops.py", line 219, in execute
    utils.getPath(context,o)
  File "/home/nemo/bin/blendercam/scripts/addons/cam/utils.py", line 3356, in getPath
    getPath3axis(context,operation)
  File "/home/nemo/bin/blendercam/scripts/addons/cam/utils.py", line 2949, in getPath3axis
    oclGetWaterline(o, chunks)
  File "/home/nemo/bin/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 176, in oclGetWaterline
    call([ os.path.join(PYTHON_BIN, bpy.utils.script_path_pref(), "addons", "cam", "opencamlib", "oclWaterline.py")])
  File "/home/nemo/bin/blender-2.78c-linux-glibc219-x86_64/2.78/python/lib/python3.5/subprocess.py", line 557, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/home/nemo/bin/blender-2.78c-linux-glibc219-x86_64/2.78/python/lib/python3.5/subprocess.py", line 947, in __init__
    restore_signals, start_new_session)
  File "/home/nemo/bin/blender-2.78c-linux-glibc219-x86_64/2.78/python/lib/python3.5/subprocess.py", line 1551, in _execute_child
    raise child_exception_type(errno_num, err_msg)
PermissionError: [Errno 13] Permission denied

location: <unknown location>:-1

location: <unknown location>:-1
```

magveda, 2017-08-24:
I am new to Blender and these are my first steps in to CAD/CAM :)

## 1. I applied the `EdgeSplit` modifier and got smooth surface with sharp edges:

[image]

Then in Edit mode I Selected Non Manifold

[image]

But I do not know what to do next with these edges :)

## 2. When I select `Pocket` strategy it cuts and removes everything through

[image]

## 3. `Profile` strategy is great for cuting out

I was adding additional plane and joining it with main object to have paths around the main object

[image]

ksuprynowicz, 2017-08-25:
Right now I have lot of work, but on Monday I'll write a tutorial on this :)

magveda, 2017-08-26:
Deal! :)

ksuprynowicz, 2017-08-28:
As promised :)

https://github.com/vilemduha/blendercam/wiki/Machining-technical-parts-using-Profile-and-Pocket-operations

Let me know if everything is clear.

closed 2023-01-14 by pppalain