vilemduha/blendercam #110: bas relief - image source not working

pppalain, 2018-02-20:
this was tried on blender 2.76.3 windows
2.77 windows
2.76 linux

Setting up image source and image target in bas relief, clicking image source opens a sort of selection pane but without anything listed
[image]
[image]

typing a filename and pressing enter does not work for image source, it works for image target.

pppalain, 2018-04-01:
anyone? hello?? is this project abandoned?

bostjannagode, 2018-04-01:
Well, im not programmer,
 im using Blendercam sucessfully for year and more. It's working as is working, for me is ok... I was struggling with  Bas relief, but on end i suceeded...
If you want to use image as z-depth source in Bas relief, you must first open image in UV-Image editor, so it can be used in Blender internally...

Best and good work,
Bostjan

[image]

pppalain, 2018-04-01:
[image]
That was my missing link - loading the image in UV editor.  Not an obvious step.  Thank you so much.

bostjannagode, 2019-03-30:
pppAlain,
it was my pleasure. 
Now i need one. What were your parameters with addon that you got that result? Somehow i dont get proper image out of there. Or even no displacement....
Thanks in advance,
Bostjan
[image]

pppalain, 2020-12-19:
basrelief was rewritten to automatically make the bumped plane.  an example bas relief blend file can be found.

closed 2020-12-19 by pppalain