vilemduha/blendercam #112: random z plunges in gcode

smithalexanderdaniel, 2018-09-01:
I've attached a blend file as well as the output for the operation "wood_roughing", which includes seemingly arbitrary plunges of the z axis down to different depths that are way way deeper than the max depth. I ran this and found out the hard way, when my machine drilled down through the spoilboard. Using Universal Gcode Sender,  was able to visualize these arbitrary plunges. Please help!

operation is an outline fill, btw.

[blend and gcode files.zip](https://github.com/vilemduha/blendercam/files/2342826/blend.and.gcode.files.zip)

ksuprynowicz, 2018-09-02:
Hi!
There's a bug in gcode export code, that shows when output precision is greater than 3 decimal places. It is caused by a value being represented in an exponential format, and then exponential being removed. It means that 0.0008 mm becomes 8 mm. I'd like to fix it, but it's hard for me to find free time now. For now, as a quick workaround I suggest using metric units (the output precision switches to 3 decimal places then).
I highly suggest verifying generated gcode with external software. In my opinion CAMotics is the best - I've avoided lots of damage by using it.

https://camotics.org/

I also suggest installing OpenCAMLib - Blendercam works way better with it. Especially the case you presented would benefit from OpenCAMLib waterline operation. Other 3D operations will also have better quality and will compute faster.

https://github.com/vilemduha/blendercam/wiki/Using-BlenderCAM-with-OpenCAMLib

smithalexanderdaniel, 2018-09-02:
Thanks,  the metric workaround helps. 

Noob question, but how do I follow the Open lib install directions? instructions are a bit cryptic... 

_install dependencies:
sudo apt install libboost-dev libomp-dev libpython-dev python-dev python-vtk6 g++ cmake git libboost-python-dev For Debian Jessie you have to replace libomp-dev with libiomp-dev (libomp-dev is not in stable yet)

Install optional dependencies:
sudo apt install amd64-microcode intel-microcode cmake-gui

Get newest OpenCAMLib (this version is patched for VTK6 compatibility):
git clone https://github.com/ksuprynowicz/opencamlib.git

If your system has VTK5, clone original OpenCAMLib instead:
git clone https://github.com/aewallin/opencamlib.git
Configure and compile:

mkdir opencamlib/build
cd opencamlib/build
cmake ..
make
Install (by default in /usr/local): sudo make install_

One other issue I'm having is that the z axis will occasionally pull all the way up to the zero height, and this results in pocket areas being only partially carved out. Effectively there are "bridges" that are unintended at arbitrary locations. See the pic below. Also it's evident in the blend file I attached above.

[image]

smithalexanderdaniel, 2018-09-02:
Ok, so I've looked through the various other issue threads, and I'm experiencing a similar problem getting OpenCAMlib and VTK working properly. I tried running python.exe (located C:\Program Files\Blender Foundation\Blender\blenderCAM64\blender-2.76\2.76\python\bin)
and entering "import vtk".

I get "no module named vtk"

I've tried downloading and installing this: vtkpython-6.3.0-Windows-64bit.exe
from here: https://www.vtk.org/download/

but when I restart python and import vtk: 
"traceback (most recent call last):
     File "<stdin>", line 1, in <module>
ImportError: No module named 'vtk'

also in blendercam, My error is still:

[image]

ksuprynowicz, 2018-09-20:
What type of operation generated unintentional bridges on the image you provided?

As for your problem, it looks like there's no python2.7 executable in executable path. Do you have working python2.7 installation?

smithalexanderdaniel, 2018-12-07:
Ok sorry to be awal for a while. I do have python 2.7 installed. I haven't resolved the issue, but for now, 

**I've got a close enough workaround:**
Don't use exact mode (make sure reducton threshold isn't too low, around .2mm works. Same for sampling raster detail, around .5mm)
Don't use opencamlib
Use waterline for multiple pass carving, NOT outline fill. This last part is critical.
Use metric, NOT imperial for units settings

I still get errors when trying to use opencamlib, but it seems I can eliminate the random Z plunges by following the steps above.

ksuprynowicz, 2019-05-08:
I've just fixed random Z plunges occurring for small Z values caused by gcode exporter. The fix is in the 2.80 beta branch.

closed 2019-05-08 by ksuprynowicz

TurBoss, 2019-06-18:
hello,

I'm just having issues with z random moves while flat surface
only on some stl faces from cad

[image]
[image]
[image]

I'll try to make a simple file that can reproduce the error

TurBoss, 2019-06-18:
here a calculated sample file

[untitled.zip](https://github.com/vilemduha/blendercam/files/3303600/untitled.zip)

Thanks

TurBoss, 2019-06-19:
it only seems to happend on imported STL mesh

ksuprynowicz, 2019-06-19:
I had the same problem. Have you tried "auto subdivide long edges"?
Also try "exact mode" and "use OpenCAMLib", it works really well for me, even with complex CAD models.

TurBoss, 2019-06-19:
thanks I'll give a try

TurBoss, 2019-06-19:
opencamlib did it smooth

Thank you so much!

ksuprynowicz, 2019-06-20:
Nice to hear that!
The problem seems to be caused by very sharp triangles. Bullet physics collision detection engine built into Blender sometimes fails to find contact point between cutter and these. This is why I initially added OpenCAMLib support. Additionally OpenCAMLib calculates toolpath up to 40 times faster on 24 core machine :)