vilemduha/blendercam #114: Windows install, position object

kenpoh01, 2019-01-12:
Hello,
Brand new to Blender CAM.  I'm a teacher and we have a Haas TM-1 at school that nobody uses since that teacher retired.  We're trying to figure out how to use it.  I've been watching the tutorials on the blogspot site and have progressed a lot in my understanding of the software.  However I have some questions.  When setting up a job, the tutorial says to place the object in the "box" representing the work area.  It then says to press the position object button to have it positioned automatically.  When I first place the object in the box then click the button, the object jumps to the center of the x y z axis instead.
First image:
[image]
After pressing "Position Object"
[image]

Also, when I try to "simulate operation", I only get a small section of the part , not the whole thing.

kenpoh01, 2019-01-12:
Any help would be appreciated.  Thanks in advance.

ksuprynowicz, 2019-01-14:
Hi!
I've just made a simple step-by-step tutorial:
https://github.com/vilemduha/blendercam/wiki/Step-by-step-tutorial
Let me know if it works for you.

closed 2023-01-14 by pppalain