vilemduha/blendercam #120: opencamlib

huntervogel, 2019-05-14:
opencamlib using blender 2.8 and blendercam-2.80 on debian stretch

file in file ~/blendercam-2.80_BETA/scripts/addons/cam/opencamlib/opencamlib.py

(around line 56)

trying to access /tmp/ocl_chunk_samples.txt before it is created.

simply creating the file /tmp/ocl_chunk_samples.txt with a few values moved the error down a bit to line 61... obviously the values where not as expected.

any ideas?

Thanks for the blender 2.8 compatible addon!

TurBoss, 2019-05-14:
check if you are using a ball cutter
in that case change it to ballnose

huntervogel, 2019-05-15:
The issue was that cmake for opencamlib was not using python2.7.

thanks for the posts clarifying that earlier.

I changed the flags pointing to anything python3 related in opencamlib/build/CMakeCache.txt then copied that folder into a fresh build director (after rm -rf * inside /build).

this post helped me figure this out: https://stackoverflow.com/questions/15291500/i-have-2-versions-of-python-installed-but-cmake-is-using-older-version-how-do

ALTERNATELY: from that post you could probably edit opencamlib/CMakeLists.txt instead  to include:
find_package( PythonInterp 2.7 REQUIRED )
find_package( PythonLibs 2.7 REQUIRED )

How do I mark this as SOLVED?

closed 2019-05-15 by huntervogel

TurBoss, 2019-05-19:
nice tnaks for your feedback