vilemduha/blendercam #129: How to add a post processor?

Precognist, 2019-10-19:
I'm trying to add a postprocessor that's not on the list. Is that doable? https://cam.autodesk.com/hsmposts?p=maslowcnc&_ga=2.257633419.1265823667.1571511427-1551755240.1571511427

Precognist, 2019-10-22:
I've found the files that act as the preprocessor settings, but they are .py files. Is there a convention for converting these? Sorry to ask so much, but the topic on the board is tapped out. (2015) https://groups.google.com/forum/embed/?place=forum%2Fblender-cam-forum#!category-topic/blender-cam-forum/development/g3Hf3lnt-6E
Blender-cam is an amazing addon and I'm having the best time using it to minimize my prototyping time!

TurBoss, 2019-10-24:
check this commits

https://github.com/vilemduha/blendercam/pull/122/files

Precognist, 2019-10-24:
Excellent. Thank you.

closed 2019-10-24 by Precognist