vilemduha/blendercam #130: Issue when creating bridges

Franckapik, 2019-12-04:
I would like to report an issue.
I use blendercam with blender 2.8 and when i try to create bridges, i have this error :

```
Traceback (most recent call last):
  File "/opt/blender/blendercam/scripts/addons/cam/ops.py", line 757, in execute
    utils.addAutoBridges(o)
  File "/opt/blender/blendercam/scripts/addons/cam/utils.py", line 2092, in addAutoBridges
    bpy.data.groups.new(bridgegroupname)
AttributeError: 'BlendData' object has no attribute 'groups'

location: <unknown location>:-1
```

Is there a way to fix this ? Thank you really really much for blendercam !!

ksuprynowicz, 2019-12-08:
Hi!
The bridges weren't ported for 2.8 yet in the version you used. I've just updated the code and they seem to work as intended now. Let mew know if they work for you. Be sure to use "master" branch now instead of "2.8_BETA" branch.

closed 2019-12-13 by ksuprynowicz

Franckapik, 2019-12-13:
Thank you really much for your help! Now it's perfect, i can use bridges with 2.8 ! :)