vilemduha/blendercam #133: G-code generation inconsistency

jmpocheau, 2019-12-12:
Hi,
First, let me congratulate you for this brilliant program.
Anyway, I think I found some kind of inconsistency.
I designed an object with Blender 2.80 and generated the tool path successfully.
Then, I mirrored this object and tried to generate the tool path, but this time, it didn't worked as planned and the result is a full removal of the entire object.
I attached the two images of the results : The first one is the successful tool path, the second is the failed one.
The main parameters are : parallel strategy, post-processing : grbl
I also attached the Blender file as a zip file.
Thanks in advance.
Jean-Marc

[TestCam.zip](https://github.com/vilemduha/blendercam/files/3955281/TestCam.zip)
[image]
[image]

TurBoss, 2019-12-13:
hello!

the random z moves is a know bug
and for what I can see about the mirror part is that is not visible when calculated
if not check to apply all transforms ( ctrl + a) and check face directions with mesh normal recalculate outsides in edit mode with all selected

pppalain, 2023-01-14:
newer versions of blender (3.14) have better collision and as far as we know, the random z has been fixed in the collision functions of blender.

closed 2023-01-14 by pppalain