vilemduha/blendercam #134: Problems without "auto subdivide long edges"

jmpocheau, 2019-12-12:
Hi, 
I found a strange issue with BlenderCam in Blender 2.80.
Sometimes the tool paths include additional "holes" which seem to be random.
I tried to analyse this and ended with a very simple example : a cube.

When I use a simple 0.2m cube as model, it works well (First screenshot).
But, when I rotate it 10°, the tool path shows strange holes inside the cube (second screenshot).
When I switch on the auto subdivide long edges parameter, everything goes right (third screenshot).

Is there any point not to keep this parameter always on ?

Thanks in advance,
Jean-Marc
[image]
[image]
[image]

TurBoss, 2019-12-13:
hello!

nice find this bug has been messing arround for a while
I'll try to find what makes it fix the random z moves

ksuprynowicz, 2019-12-13:
Hi!
The problem with random z plunges is caused by collision detection library which is built into Blender.
Currently there are two ways to fix it - one is, as you said, selecting auto subdivide long edges option (be careful, it may still happen sometimes so still run simulation before machining).
The second way is using OpenCAMLib for collision detection:
https://github.com/vilemduha/blendercam/wiki/Using-BlenderCAM-with-OpenCAMLib
OpenCAMLib is very fast (up to 50 times faster on 24-core machine) and always gives accurate results.

>  Is there any point not to keep this parameter always on ?

The reason that "auto subdivide long edges" is not always set is that it would cause problems like increased memory usage for very complex models (like 10+ million polygons), but I think that automatically switching it on when using collision detection integrated in Blender would be a good idea (correct results are more important than performance), as long as it is always off when using OpenCAMLib.

jmpocheau, 2019-12-13:
Thanks a lot for those explanations.
I will try to give OpenCAMLib a new chance ... I installed it but only got error message.
I'll review my installation and let you know if I find any problem.
Thanks again

closed 2019-12-13 by jmpocheau