vilemduha/blendercam #136: calculate path for groups

TurBoss, 2019-12-16:
there is some un ported code that still uses groups and not collections

tried to fix but can't figure out

Franckapik, 2019-12-18:
It seem's that we have the same when we try to use drill strategy ... 

[image]

ksuprynowicz, 2019-12-19:
I'll have a look at it soon, collection as an operation source would be useful and drill is a must have.

Franckapik, 2020-01-09:
Is there a way we can help on this ? Is this just a kind of replacement of words or we have to rewrite totally the code ?

ksuprynowicz, 2020-01-09:
Thank you for reminding me about this. Luckily it turned out to be mostly word replacements, with one rather interesting bug - endless duplication of objects :)
I've tested it on collections of meshes and collections of curves and it seems to work. I still have to test it with OpenCAMLib.
Let me know if it works for you.

ksuprynowicz, 2020-01-09:
Drill strategy also seems to work again as intended.

TurBoss, 2020-01-09:
I'm just testing it
amazing work 
thank you so much

Franckapik, 2020-01-10:
It works !
Thank you so much for having spending time for this !! 
BlenderCam is a excellent tool now !

ksuprynowicz, 2020-01-10:
You're welcome! :)

closed 2020-01-10 by TurBoss