vilemduha/blendercam #137: Missing Panel Buttons

sydobrew, 2020-01-17:
Hello, I just recently installed Blender CAM via github installation instructions using Blender 2.79 on Ubuntu 18.04.  Everything seemd to install properly but from the tutorial screen shots is shows there should be buttons to the side of the CAM Operation panel to create a new operation.  These buttons seem to be missing.  Any help would be greatly appreciated, I am very interested to explore this software package.

ksuprynowicz, 2020-01-17:
Hi!
"Master" branch is not compatible anymore with Blender 2.79. Can you try with 2.80?

sydobrew, 2020-01-17:
Hello thank you for the feedback.  I have tried to install 2.8 but the blender website redirects 2.8 downloads to 2.81a.  I downloaded 2.81a and ran it from the directory, there doesn't seem to be an install option.  This new blender package seems to use a bundled version of python instead of the system python.  This leaves a dependency for python module shapely unsatisfied.  I have tried running the blender package under the standard system python with shapely installed and I created a virtualenv with shapely installed both setups failed to find the shapely package from within blender.  Again any direction from here would be greatly appreciated.

ksuprynowicz, 2020-01-17:
I think it's worth trying with 2.81a, if it won't work let me know I'll update the code on tuesday.
With the bundled python you can install shapely with PIP. Here's how to do it:

https://github.com/vilemduha/blendercam/wiki/Using-Blendercam-from-github#dependencies

The only difference is that for 2.81 instead of `cd 2.80/python/bin/` you'll probably have `cd 2.81/python/bin/` .
Let me know if it works for you.

sydobrew, 2020-01-17:
Everything seems to be working now with 2.81a and the dependency install.  Thank you for your help.

closed 2020-01-17 by ksuprynowicz