vilemduha/blendercam #138: Cannot enable CAM: Shapely not available (but it is)

Silventar, 2020-01-27:
Attempting to enable BlenderCAM in the addon window and I get told that shapely is not available. But it is...... See attached error screen.

[image]

Silventar, 2020-01-27:
I should note that I am using Blender 2.81a.

TurBoss, 2020-01-28:
how did you install shapely on windows?

Silventar, 2020-01-28:
Evening! I installed it with GIT BASH while I was I. The /Blender Foundation/ Blender 2.81/2.81/python/bin folder. I had to install with wheels from the download as the command on the Blendercam git page would not work. I also ahd to add - - user in order for anything to work.

Silventar, 2020-01-28:
Basically, I had to use this method: https://towardsdatascience.com/install-shapely-on-windows-72b6581bb46c

but with the newer version of Shapely and from within the folder mentioned.

Silventar, 2020-01-28:
silve@Audio-Laptop MINGW64 /c/Program Files/Blender Foundation/Blender 2.81/2.81/python/bin
$ python -m pip install C:/Users/silve/Downloads/Shapely-1.6.4.post2-cp38-cp38-win32.whl --user
Requirement already satisfied: Shapely==1.6.4.post2 from file:///C:/Users/silve/Downloads/Shapely-1.6.4.post2-cp38-cp38-win32.whl in c:\users\silve\appdata\roaming\python\python38\site-packages (1.6.4.post2)

Silventar, 2020-01-28:
Ok, so   lender uses its own python install and not the system install (that seems dumb) and shapely I stalled into the system python. I don't know how to install it into blenders custom nonsense.  The versions are different; blender has an older version of python. How do I install shapely onto m lender python on win10?

TurBoss, 2020-01-28:
could you try this from the blender python bin direcory?

./python.exe -m pip install C:/Users/silve/Downloads/Shapely-1.6.4.post2-cp38-cp38-win32.whl --user

Silventar, 2020-01-28:
PS C:\Program Files\Blender Foundation\Blender 2.81\2.81\python\bin> python.exe -m pip install C:/Users/silve/Downloads/Shapely-1.6.4.post2-cp38-cp38-win32.whl --user
Requirement already satisfied: Shapely==1.6.4.post2 from file:///C:/Users/silve/Downloads/Shapely-1.6.4.post2-cp38-cp38-win32.whl in c:\program files (x86)\python38-32\lib\site-packages (1.6.4.post2)

Silventar, 2020-01-28:
Sorry, the above was from powershell.

The below is from Git BASH:

silve@Audio-Laptop MINGW64 /c/Program Files/Blender Foundation/Blender 2.81/2.81/python/bin
$ PS C:\Program Files\Blender Foundation\Blender 2.81\2.81\python\bin>./python.exe -m pip install C:/Users/silve/Downloads/Shapely-1.6.4.post2-cp38-cp38-win32.whl --user
bash: ./python.exe: Permission denied

TurBoss, 2020-01-28:
its installing on the sistem python 38

it must be installed on the blender bundled directory

try to add the "./" at the begining of the command and remove the --user option

Silventar, 2020-01-28:
I have a headache now. LOL!
I ran as suggested:

PS C:\Program Files\Blender Foundation\Blender 2.81\2.81\python\bin> ./python.exe -m pip install C:/Users/silve/Downloads/Shapely-1.6.4.post2-cp38-cp38-win32.whl
ERROR: Shapely-1.6.4.post2-cp38-cp38-win32.whl is not a supported wheel on this platform.

Argh. Then I remembered that my system python is 3.8, but BLENDER is 3.7. They are a version BEHIND. So I downloaded Python 3.7.6 and installed it on my system..which means I have THREE VERSIONS OF PYTHON on my system now.  I added it to the PATH variable. Then I reran the code as you suggested, but used a different wheel. NOW it works. Blendercam is installed! sheesh.

Here is the final output that worked.

PS C:\Program Files\Blender Foundation\Blender 2.81\2.81\python\bin> ./python.exe -m pip install C:/Users/silve/Downloads/Shapely-1.6.4.post2-cp37-cp37m-win_amd64.whl
Processing c:\users\silve\downloads\shapely-1.6.4.post2-cp37-cp37m-win_amd64.whl
Installing collected packages: Shapely
Successfully installed Shapely-1.6.4.post2

What was odd before was that my architecture is 64bit, but the origional 3.8 system python was 32. Now it is all sorted out...except for my headache. hehe!

TurBoss, thank you so much for your help, I really appreciate it!

TurBoss, 2020-01-28:
you  are welcome!

closed 2020-05-15 by ksuprynowicz