vilemduha/blendercam #139: Impossible to use Parrallel, Carve, Cross, ... strategies...

Franckapik, 2020-02-12:
I use BlenderCam for Cutout and Pocket but i have always this error when i try others strategies : 

Traceback (most recent call last):
  File "/opt/blender/blendercam/scripts/addons/cam/ops.py", line 234, in execute
    utils.getPath(context, o)
  File "/opt/blender/blendercam/scripts/addons/cam/utils.py", line 3452, in getPath
    getPath3axis(context, operation)
  File "/opt/blender/blendercam/scripts/addons/cam/utils.py", line 3011, in getPath3axis
    chunks.extend(sampleChunks(o, pathSamples, layers))
  File "/opt/blender/blendercam/scripts/addons/cam/utils.py", line 391, in sampleChunks
    prepareArea(o)
  File "/opt/blender/blendercam/scripts/addons/cam/image_utils.py", line 1792, in prepareArea
    renderSampleImage(o)
  File "/opt/blender/blendercam/scripts/addons/cam/image_utils.py", line 1653, in renderSampleImage
    n.links.new(n1.outputs['Z'], n2.inputs['Image'])
KeyError: 'bpy_prop_collection[key]: key "Z" not found'

location: <unknown location>:-1

Is there something i do wrong ??

Thank you a lot !!

ksuprynowicz, 2020-02-12:
Do you have "Use exact mode" option selected?

Franckapik, 2020-02-12:
Thank you ksuprynowicz! It's working if i select the use excat mode option and if i put my object on Scene Collection only without new Collection. 

Than you a lot for your work and your answers !

ksuprynowicz, 2020-02-12:
You're welcome!
It should also work if there are more collections in the scene - I use as many as 30 for complex projects with multiple machined parts. Maybe it depends on which collection is active?

The "Use exact mode" option is there because earlier on BlenderCAM had also a "Raster Mode" that could source data either from rendered depth map or image. Depth map was rendered using Blender Internal renderer that was removed in 2.8, so raster mode needs to be rewritten to use Eevee renderer. I'm using mostly exact mode + OpenCAMLib, because its fastest and  gives best results.

closed 2020-05-15 by ksuprynowicz