vilemduha/blendercam #144: no such file or directory error

ilya-11, 2020-05-04:
Good day to everyone!

I am sorry if this is an obvious question, but I am no programmer, and new to the whole CAM/milling topic all together: It just so happened I like Linux (running Debian 9 right now), so I try to use Linux software where ever I can.

I have installed latest stable Blender (downloaded, not from Debian repository) and BlenderCAM as described here, read the Tutorial and trying to get a 3-axis G-code for a simple half/cylinder object. But unable to do that due to persisting "no such file or directory" error referring to file '/tmp/ocl-chunk-samples.txt' file.
And indeed I can't find such file in '/tmp' directory !
Should I generate it somehow preliminary to path calculation?

I tried to play with different settings, including "use exact mode" with openCAMlib - no luck so far!

Would really appriciate the help!
Thanks in advance!

ilya-11, 2020-05-04:
Played around further: seems like the error is on the side of openCAMlib..
But still no solution how to deal with it rather then disable it...

ksuprynowicz, 2020-05-12:
Hi!
To select "use opencamlib" option you need to have opencamlib installed, but the installation procedure will change soon (it'll be easier to use). Use exact mode should work fine without opencamlib. If you still can't solve the problem, you can post example file here, so I'll take a look at it.

ksuprynowicz, 2020-05-12:
Another problem is that Waterline operation stopped working at all, and I get following error message:
`NameError: name 'operationSettingsToCSV' is not defined`

ilya-11, 2020-05-12:
Hey!
Thank you very much for the response!

Yes, without opencamlib everything seems to work, so I probably did
something wrong during opencamlib installation.
I will work then in this mode until this new installation method you have
mentioned arrives for opencamlib.

Thanks again for your help and for the hard work with this BlenderCAM
extension!
With the guys like you one day hopefully we will be able to do all kind of
things in Linux with opensource software!

Have a good day!

Regards,

Ilya.

ksuprynowicz, 2020-05-15:
Ok, it is updated now :)
Take a look at these pages:
https://github.com/vilemduha/blendercam/wiki/Install-opencamlib-into-blender-2.80-on-linux
https://github.com/vilemduha/blendercam/wiki/Using-BlenderCAM-with-OpenCAMLib
Be sure to set option "Show experimental features" in Blender User Preferences / Addons / CAM - gcode generation tools - high quality waterline strategy will be available.
Let me know it if works for you.

ilya-11, 2020-05-19:
Thank you very much!!

I'll try to make a new installation according to the instruction shortly
and let you know if it worked!

Have a good day!

ilya-11, 2020-05-19:
During installation

You are using pip version 19.0.3, however version 20.1.1 is available.

And proposition to upgrade.

Should I upgrade or use the older version, I wonder?? Is it important for
the case??

Thank you!

ilya-11, 2020-05-19:
bumped into this thing

CMake Error at CMakeLists.txt:2 (cmake_minimum_required):
  CMake 3.9 or higher is required.  You are running version 3.7.2

-- Configuring incomplete, errors occurred!

I run Debian 9 Stretch, fully upgraded, but guess they havn't added latest
Cmake to repositories yet..
Have to find a way to upgrade it from different source I guess..

ksuprynowicz, 2020-05-24:
There are two options for this:
- Opencamlib probably doesn't actually require cmake 3.9, you can try to lower required version in CMakeLists.txt - relpace 3.9 requirement with: `cmake_minimum_required(VERSION 3.7.2 FATAL_ERROR)`
- Add Debian Testing repository to your apt sources with extremely low priority and install cmake 3.9.

ilya-11, 2020-05-26:
Thank you, mate!!
I'll try this on weekend and let you know the results!

ilya-11, 2020-05-26:
But even now I was able to cut my first relief with spiral tool path made
by your CAM - and I it was really nice!
Just wanted to thank you again for the great job with this extension!!

ksuprynowicz, 2020-05-26:
Nice to hear that!
The extension is not mine though, it was originally written by Vilém Duha, and now supporting it is a team effort. Every contribution is welcome :)

ilya-11, 2020-05-28:
Thanks for clarification of credits!
I'll find a way to make at least a small contribution to this great
project!)

closed 2023-01-14 by pppalain