vilemduha/blendercam #145: Python 3.8 removes deprecated time.clock, use time.perf_counter or time.process_time

boxofrox, 2020-05-08:
ArchLinux has migrated to Python 3.8 which removes the `time.clock` method, resulting in the following error when attempting to calculate a cam path.

```
bpy.ops.object.calculate_cam_path()
Traceback (most recent call last):
  File "/usr/share/blender/2.82/scripts/addons/cam/ops.py", line 234, in execute
    utils.getPath(context, o)
  File "/usr/share/blender/2.82/scripts/addons/cam/utils.py", line 3425, in getPath
    t = time.clock()
AttributeError: module 'time' has no attribute 'clock'

location: <unknown location>:-1
```

`time.clock` is considered deprecated since Python 3.3, and with Python 2.x EOL'd, I'm curious what direction blendercam intends given the readme still lists Python 2.7 as a requirement.

I'm happy to attempt a PR, but I'm not a Python programmer and unfamiliar with any process for writing scripts compatible with Python 2 & 3, if that's an intended goal of this project.

Blender: 2.82.a
BlenderCAM: git via AUR blendercam-git
Python: 3.8.2 (per the scripting workspace in Blender)

boxofrox, 2020-05-08:
Doh, PR #143 addresses this.  Thought I checked the PRs before writing submitting this.

Sorry for the noise.

closed 2020-05-08 by boxofrox

ksuprynowicz, 2020-05-12:
No problem, I should have merged it sooner :)