vilemduha/blendercam #150: I need help.(G-code)

mir132, 2020-05-14:
I have this file. 
It is not a simple 2D object.
 I want to draw its surface outline and get a path to fill the inside. 
How can I do this?
I want to get the path as shown in the picture.
[image]
[image]

ksuprynowicz, 2020-05-15:
Use Parallel operation. You can set the angle of paths with "Angle of paths" option.
To limit cutter movement like you shown above you select "Around" in the option "Ambient". If you want to keep the cutter completely inside, use option "Cutter stays in ambient limits".
Let me know if it works for you.

mir132, 2020-05-15:
@ksuprynowicz 
[image]

[image]
Thanks for your answer. I ran the 2.76 blender downloaded from the Google driver. And only the sample file downloadable from the homepage could get the path of the surface. However, I couldn't get the path by creating a new page in Blender. What options or settings should I add? I am currently using window10. I want to use 2.76blender if possible.

ksuprynowicz, 2020-05-15:
The version you want to use is already 4 years old. I advise against using it because lots of problems were fixed in between. You can try selecting "Cam" as renderer and switching on "Use exact mode", if there's such option in that release.

closed 2020-05-24 by ksuprynowicz