vilemduha/blendercam #151: I want to install BLENDERCAM on Blender 2.79. It is installed by WINDOWPC. and  an error occurs when the stl file is loaded and proceeds without opening the sample file.

mir132, 2020-05-15:
I want to install BLENDERCAM on Blender 2.79. It is installed by WINDOWPC.
I want to know the detailed method.
And when working with Blender 2.76 downloaded from Blender Google Drive, when the sample file is opened and an arbitrary stl file is loaded, the work proceeds. However, an error occurs when the stl file is loaded and proceeds without opening the sample file. I want to know what the reason is.
Please answer.
Thank you.

ksuprynowicz, 2020-05-15:
Blender 2.79 is not supported anymore, and that 2.76 downloadable distribution is very outdated.
You should download Blender 2.82a and install Blendercam as described:
https://github.com/vilemduha/blendercam/wiki/Using-Blendercam-from-github

mir132, 2020-05-18:
@ksuprynowicz 
[image]
[image]
I am not sure why it is not installed.
I want a solution.

ksuprynowicz, 2020-05-18:
Shapely is not available because you are installing it on system python installation.
Each Blender version has it's own bundled python installation, so in the command line you have to go to your blender installation directory (using cd command), then to directory with bundled Python - "2.82/python/bin/",
and then execute that version of python:
https://github.com/vilemduha/blendercam/wiki/Using-Blendercam-from-github
You also should use Blender 2.82a - current version will not work with 2.79 at all due to Blender API changes.
Let me know if this works for you.

mir132, 2020-05-18:
Thanks for the answer.
I'm going to try applying it to the 2.82a blender.
After saving the Blender file, I was able to proceed. Thank you.
I have a question.
I want to move the worktable to the target and execute the work.
In other words, I want to proceed without changing the location of the stl file and maintaining the original coordinates.
As shown in the picture, the file I want is far from the workbench.
I want to follow the coordinates of the stl file. If you move the position of stl it will break it.
[image]

I am very grateful for each kind answer. Thank you.

ksuprynowicz, 2020-05-18:
You're welcome!
To preserve original coordinates you should put 3d cursor at world origin, import STL, change pivot point to 3d cursor (this is very important, otherwise model will move and coordinates won't match), and scale imported model by 1/1000 if it is in millimetre unit scale.
When you run the gcode on CNC machine, you should first do a test run with coordinate system placed a bit higher, so you won't ruin the workpiece if something went wrong.

mir132, 2020-05-18:
@ksuprynowicz 
Thanks for the answer, sorry. I don't seem to understand your answer. I'm sorry, can you explain in detail?
As in the picture,
In the left state, no path is created. So I want to move the workbench to the right, but the Z value is negative. I want to get G-CODE that follows STL's XYZ. I expected G-CODE to be made like a white line, but it wasn't. I want the G-CODE to be generated like a white line.
[image]
OR
[image]

ksuprynowicz, 2020-05-18:
You don't have to move machine area for that. You just have to set "Free movement height" to safe level above your object, and "Operation depth start" to a height above your object. If you also want to set initial and final cutter position a bit higher, set "Use position definition checkbox" and enter coordinates there.
I've just tested it and seems to work fine:
[image]

mir132, 2020-05-19:
@ksuprynowicz 

Thank you for your kindness. 
Is the above explanation only available in version 2.82?
Additionally, I want to install 2.82a blendercam on a window pc. However, an error occurs because the description is a Linux version. I want to know the solution. And I want to know the meaning of abc when using 5 or 6 axis. Sorry to send many questions every time.

[image]
[image]

ksuprynowicz, 2020-05-24:
It seems like you have to do `pip3 install shapely` again after installing wheel.
As for 5 and 6-axis support, it's unfinished right now. I have a 6-axis machine, but it needs new electronics so I don't have any experience with it.

mir132, 2020-05-28:
@ksuprynowicz 
Thanks for the answer.
There are additional questions. On the surface
When making a path with an OUTLINEFILL, floating occurs without touching it as shown in the figure.
How can I solve it?

And is there any way to make Gcode draw in the order I want?
For example, how to draw in order from the left when writing letters, and how to fill the inside in parallel after drawing the outline?

[image]

[image]

ksuprynowicz, 2020-05-28:
That's a bit surprising. Can you post a simple example file where this occurs?

As for enforcing order, right now you have to create separate operations for that.

mir132, 2020-05-29:
[test3_0529.zip](https://github.com/vilemduha/blendercam/files/4699250/test3_0529.zip)

I wrapped the svg file on a curved surface.
Then I created a path. The result is the same as the file. Why?
Thanks for your kind answer every time.

pppalain, 2023-01-14:
issue closed as blender 2.79 is unsuported with new code.

closed 2023-01-14 by pppalain