vilemduha/blendercam #152: Using BlenderCam with layers ?

Franckapik, 2020-05-15:
[image]

Hi!
Because i have multiples versions of my work, i was thinking of using multiples layers.
So i did a full copy and change some size of mesh, and then it seem's that i can't do that with this error :

`Traceback (most recent call last):
  File "/opt/blender/blendercam/scripts/addons/cam/ops.py", line 320, in execute
    bpy.ops.object.calculate_cam_path()
  File "/opt/blender/2.80/scripts/modules/bpy/ops.py", line 201, in __call__
    ret = op_call(self.idname_py(), None, kw)
RuntimeError: Error: Traceback (most recent call last):
  File "/opt/blender/blendercam/scripts/addons/cam/ops.py", line 216, in execute
    path.select_set(state=True)
RuntimeError: Error: Object 'cam_path_Vis' can't be selected because it is not in View Layer 'RenderLayer'!

location: /opt/blender/2.80/scripts/modules/bpy/ops.py:201

location: <unknown location>:-1

Traceback (most recent call last):
  File "/opt/blender/blendercam/scripts/addons/cam/ops.py", line 320, in execute
    bpy.ops.object.calculate_cam_path()
  File "/opt/blender/2.80/scripts/modules/bpy/ops.py", line 201, in __call__
    ret = op_call(self.idname_py(), None, kw)
RuntimeError: Error: Traceback (most recent call last):
  File "/opt/blender/blendercam/scripts/addons/cam/ops.py", line 216, in execute
    path.select_set(state=True)
RuntimeError: Error: Object 'cam_path_Vis' can't be selected because it is not in View Layer 'RenderLayer'!

location: /opt/blender/2.80/scripts/modules/bpy/ops.py:201

location: <unknown location>:-1
`

What do you think about this ? Is there something i can do ? Thank you again for BlenderCam, i use it everyday !!

Franckapik, 2020-05-15:
Oh i see, when i copy a scene, all meshs are copied also with .001 but obviously not in cam operation so it didn't retrieve the mesh. I should maybe organise this in a different way maybe with differents layers... If you have some advices on this ? ;) It's finally a blender question sorry...

ksuprynowicz, 2020-05-15:
I typically do it with collections instead of scenes - in that way everything can be organized hierarchically. Currently BlenderCAM can't reach objects that are hidden or are in different scene, but that would be easy to fix - if the object cannot be selected it should be shown if it is on visible layer, and if it is on hidden layer or other scene it should be temporarily linked to active collection and unlinked again after computing toolpath. If you know a bit of Python scripting, you could submit a patch, it would help me too :)

closed 2021-04-29 by Franckapik