vilemduha/blendercam #154: Invalidity of geometry when import dxf

Franckapik, 2020-05-27:
Hi!

I've tried to import different dxf and then use Cutout path but i have always the same kind of error : 

`Traceback (most recent call last):
  File "/opt/blender/blendercam/scripts/addons/cam/chunk.py", line 757, in chunksToShapely
    ch.poly)  # sgeometry.Polygon( ch.parents[0].poly, ch.poly)
  File "/opt/blender/2.80/python/lib/python3.7/site-packages/shapely/geometry/base.py", line 616, in difference
    return geom_factory(self.impl['difference'](self, other))
  File "/opt/blender/2.80/python/lib/python3.7/site-packages/shapely/topology.py", line 70, in __call__
    self._check_topology(err, this, other)
  File "/opt/blender/2.80/python/lib/python3.7/site-packages/shapely/topology.py", line 38, in _check_topology
    self.fn.__name__, repr(geom)))
shapely.errors.TopologicalError: The operation 'GEOSDifference_r' could not be performed. Likely cause is invalidity of the geometry <shapely.geometry.polygon.Polygon object at 0x7f8063ad1da0>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/blender/blendercam/scripts/addons/cam/chunk.py", line 810, in chunksToShapely
    ch.poly)  # sgeometry.Polygon( ch.parents[0].poly, ch.poly)
  File "/opt/blender/2.80/python/lib/python3.7/site-packages/shapely/geometry/base.py", line 616, in difference
    return geom_factory(self.impl['difference'](self, other))
  File "/opt/blender/2.80/python/lib/python3.7/site-packages/shapely/topology.py", line 70, in __call__
    self._check_topology(err, this, other)
  File "/opt/blender/2.80/python/lib/python3.7/site-packages/shapely/topology.py", line 38, in _check_topology
    self.fn.__name__, repr(geom)))
shapely.errors.TopologicalError: The operation 'GEOSDifference_r' could not be performed. Likely cause is invalidity of the geometry <shapely.geometry.polygon.Polygon object at 0x7f8063ad1da0>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/blender/blendercam/scripts/addons/cam/ops.py", line 234, in execute
    utils.getPath(context, o)
  File "/opt/blender/blendercam/scripts/addons/cam/utils.py", line 3452, in getPath
    getPath3axis(context, operation)
  File "/opt/blender/blendercam/scripts/addons/cam/utils.py", line 2948, in getPath3axis
    strategy_cutout(o)
  File "/opt/blender/blendercam/scripts/addons/cam/utils.py", line 2317, in strategy_cutout
    p = getObjectOutline(o.cutter_diameter / 2, o, offset)
  File "/opt/blender/blendercam/scripts/addons/cam/utils.py", line 1885, in getObjectOutline
    polygons = getOperationSilhouete(o)
  File "/opt/blender/blendercam/scripts/addons/cam/utils.py", line 1764, in getOperationSilhouete
    operation.silhouete = getObjectSilhouete(stype, objects=operation.objects)
  File "/opt/blender/blendercam/scripts/addons/cam/utils.py", line 1777, in getObjectSilhouete
    silhouete = chunksToShapely(allchunks)
  File "/opt/blender/blendercam/scripts/addons/cam/chunk.py", line 861, in chunksToShapely
    ch.poly)  # sgeometry.Polygon( ch.parents[0].poly, ch.poly)
  File "/opt/blender/2.80/python/lib/python3.7/site-packages/shapely/geometry/base.py", line 616, in difference
    return geom_factory(self.impl['difference'](self, other))
  File "/opt/blender/2.80/python/lib/python3.7/site-packages/shapely/topology.py", line 70, in __call__
    self._check_topology(err, this, other)
  File "/opt/blender/2.80/python/lib/python3.7/site-packages/shapely/topology.py", line 38, in _check_topology
    self.fn.__name__, repr(geom)))
shapely.errors.TopologicalError: The operation 'GEOSDifference_r' could not be performed. Likely cause is invalidity of the geometry <shapely.geometry.polygon.Polygon object at 0x7f80608069e8>

location: <unknown location>:-1
`

Is there something to do before on the curve (i try to remove doubles) to avoid this error ? 

Thank you !!

[image]

ksuprynowicz, 2020-05-28:
I've encountered this problem, and it seems that it is due to Z scale in imported dxf geometry being set to 0. Try changing scale to 1 for each object and let me know if it works.

Franckapik, 2020-05-28:
The Z scale on blender is to 1. So you mean that i should change z scale in dxf file in another tool than blender ? I try to find this option in librecad but without success ... :/

ksuprynowicz, 2020-05-28:
Only other thing that could cause this would be non-continous or self intersecting curves. Then the best way to fix this is to convert curve to mesh, then use "merge by distance" and convert it to curve again when it is continous and closed. If this won't help then there may be a bug in BlenderCAM. Can you post a simple example where this occurs?

Franckapik, 2020-05-28:
Thank you ! 
The other way with the "merge by distance" work with one of the three dxf imported example.
I believe that it's probably of issue with non-continous and self intersecting curves. I think this is finally for laser cut so it's not a good idea to use Cutout (Outside) with this kind of draw..
Thank you really much for all your help !!

closed 2020-05-28 by Franckapik