vilemduha/blendercam #155: ModulenotFoundError: No module named ocl

midoec, 2020-05-28:
I just downloaded blendercam and tried to set it up on ubuntu 18.04LTS with blender 2.82a and blender 2.81a they are both giving this error. ModulenotFoundError: No module named ocl

There is a copy of blendercam i downloaded in January. That copy works with blender 2.81a and still works now without Opencamlib. I just wanted to get the latest code and i dont know why its giving the error.

ksuprynowicz, 2020-05-28:
Thanks for reporting this. I've just added a fix for this, let me know if it works.

midoec, 2020-05-28:
hello Thanks for fixing it. it works now.

I am really grateful for this addon and i cant thank you guys enough. i have been using it since january and its simply amazing. Thanks once again.

ksuprynowicz, 2020-05-28:
Nice to hear that :)
Have you checked "show experimental features" in addon preferences? There's more options this way, some are unfinished, but some are very useful.

closed 2020-05-28 by ksuprynowicz

alexpGH, 2021-01-26:
HI all,
I still face the above problem with the current master version (fd61ffe 14 days ago).
Did the fix get lost?