vilemduha/blendercam #156: Issues with 3d strategies

alexpGH, 2020-05-29:
Hi,
to separate the topics I open this separate topic for the issues I have with 3d strategies.
**Aspect 1**
The good news is that if I use parallel strategy on the cube with inset [blend file](https://blenderartists.org/uploads/short-url/57oePeLMSx34nHbTxuT3tIrBW9U.blend) with 
- exact mode
- opencamlib / no opencamlib (so its not a problem of opencamlib)
- operation depth end less deep than the pocket depth 

the result look reasonable: the inset depth is -2cm indicated by the curve and the depth limit was set to -1cm
[image]

the same works for -0.5cm.
However, if I select _depth from object_ I would expect the milling going down to the inset depth, which is not the case. It seems the same milling depth that has been explicitly given is applied.
If I e.g. set it to -0.5cm, then select depth from object and calculate path, I get exactly -0.5cm milling depth!
The same happens if I set the explicit depth to 0.0m; in this case, the last explicit value (e.g. -0.5cm) is used.
I ensured that it really a new cam path by deleting the old cam path before recalculating the new path.

**Aspect 2**
If I set the explicit depth to exactly the depth of the inset (-2cm) or even below, I get (independent of opencamllib usage):

`Traceback (most recent call last):
  File "/home/alexp/wicht_soft/blendercam/scripts/addons/cam/ops.py", line 234, in execute
    utils.getPath(context, o)
  File "/home/alexp/wicht_soft/blendercam/scripts/addons/cam/utils.py", line 3457, in getPath
    getPath3axis(context, operation)
  File "/home/alexp/wicht_soft/blendercam/scripts/addons/cam/utils.py", line 3016, in getPath3axis
    chunks.extend(sampleChunks(o, pathSamples, layers))
  File "/home/alexp/wicht_soft/blendercam/scripts/addons/cam/utils.py", line 529, in sampleChunks
    ratio = (splitz - v1.z) / (v2.z - v1.z)
ZeroDivisionError: float division by zero
` 

Deselecting _exact mode_ in any case (also for sensible explicit depth) gives an error 
`location: <unknown location>:-1
Error: Traceback (most recent call last):
  File "/home/alexp/wicht_soft/blendercam/scripts/addons/cam/ops.py", line 234, in execute
    utils.getPath(context, o)
  File "/home/alexp/wicht_soft/blendercam/scripts/addons/cam/utils.py", line 3457, in getPath
    getPath3axis(context, operation)
  File "/home/alexp/wicht_soft/blendercam/scripts/addons/cam/utils.py", line 3016, in getPath3axis
    chunks.extend(sampleChunks(o, pathSamples, layers))
  File "/home/alexp/wicht_soft/blendercam/scripts/addons/cam/utils.py", line 391, in sampleChunks
    prepareArea(o)
  File "/home/alexp/wicht_soft/blendercam/scripts/addons/cam/image_utils.py", line 1792, in prepareArea
    renderSampleImage(o)
  File "/home/alexp/wicht_soft/blendercam/scripts/addons/cam/image_utils.py", line 1653, in renderSampleImage
    n.links.new(n1.outputs['Z'], n2.inputs['Image'])
KeyError: 'bpy_prop_collection[key]: key "Z" not found'

location: <unknown location>:-1
`
This last one is no problem as I anyhow would use exact mode.

Continued... next for using a bowl

alexpGH, 2020-05-29:
Note: using material size: estimate from model solved the issue for the pocket operation.
Therefore I tried using it here too.
In this case, selecting _depth from object_ (together with cam material estimate from model)
results in the same ZeroDivisionError reported in the first post

alexpGH, 2020-05-29:
Now for curved surfaces. using a bowl and applying parallel with

- exact mode
- opencamlib
- explicit material size
- explicit depth

works reasonable:
[image]
However, skin setting is also ignored (also with cam material estimated from model), which is a pity -> no rouhening passes for curved surfaces.

ksuprynowicz, 2020-05-29:
The exact mode should be always on in Blender 2.8 and newer. If it can't be fixed it would be best to remove that option. Division by zero seems to be caused by explicitly stating material size - if you switch to "estimate from model", it should work.
A temporary fix for lack of Skin option is to move object up by skin value and increase cutter diameter by 2*skin value.

alexpGH, 2020-06-29:
I tested the first case above (the square pocket in the cube) also with material size: estimate from model: still the depth from object does not work. As described above, the last explicit value used is applied also in this case.

alexpGH, 2020-06-30:
Additional information:
Using explicit material size & or explicit depth works well with the bowwl with inset.
-> I guess the division by zero error for the square pocket must be a subsequent fault related to the planar pocket...

ksuprynowicz, 2020-07-23:
I'm sorry that I didn't replied earlier - I had a lot of work this month. Can you post a blend file with both successful and failed operation so I could take a look at it?

pppalain, 2021-03-12:
are these issues still alive?  If so, join the blendercam room in element we'll tackle these

closed 2023-01-14 by pppalain