vilemduha/blendercam #158: Improvement - Add a copy paste button on CAM chain like for the CAM operation

Franckapik, 2020-06-07:
Hello,
When for example, we use Cutout, it's usefull to make one chain operation with screw drill and another one without. The same with a mesh added for testing before cuting all the others.
It could be really usefull to copy paste chain operation and then adding/removing some cam operation to have more flexibility on milling.

I just report a need, but maybe i am the only one to do this ;)
Again, thank you really much for all your work :)

F.

ksuprynowicz, 2020-06-07:
Hi!
That would be a welcome change, but it wouldn't work for when you want to have different input objects for copied chain. Current chains and operations have to be copied by hand (which is time consuming) if you have many similar objects using the same or nearly same technological process but different input objects and output files. I want to add a node system, so you would have to create operation only once and then reuse it in different chains with different datasets. The chain would then be represented by a node system, and it would be very easy to copy it and replace input objects. What do you think about this?

Franckapik, 2020-06-07:
It would be so nice to have a node system to manage cam/chain operation! 
You were thinking about using existing node panel from blender ? (Is that possible !?)

ksuprynowicz, 2020-06-07:
Sure! Luxcorerender uses custom nodes panels, just like the one for materials in Cycles. I'll probably have a look at the code and try to do something similar.

Franckapik, 2020-10-15:
I understand that it would be a hudge work on it but have you got a look on a node system for blendercam? 
I would like so much learn python to contribute but I think that I have a long way to do this... 😐

ksuprynowicz, 2020-10-15:
Sadly I probably won't be able to do it for now. My health is failing and I will have to close my CNC machining company because of that. I will still work on BlenderCAM (I'm keeping the CNC mill to do some hobby work on it), but probably nothing big until I get back to health.

abosafia, 2020-10-15:
I wish you get back to health soon
Thanks a lot for helping us

Franckapik, 2020-10-15:
I will say the same! Thank you a lot for all your help hoping you get back to health soon!!

closed 2021-04-29 by Franckapik