vilemduha/blendercam #159: how about pycam and FreeCAD/path/CAM ?

lalberts, 2020-07-23:
Hello,
isn't there a lot of common good ?
is there now way to concentrate the efforts of each group so that one can profit from the other?

how far appart are they?

Lukas

ksuprynowicz, 2020-07-23:
There definitely is. A lot of code is shared already, because both FreeCAD and BlenderCAM use opencamlib.
FreeCAD probably couldn't use BlenderCAM code directly due to a license incompatibility (MIT vs. GPLv2).
BlenderCAM cant use pycam code because pycam has later GPL license (GPLv3).
It is questionable whether pycam can use BlenderCAM code, because it if not specified if BlenderCAM can be distributed on later GPL versions.
Personally I'm all for releasing BlenderCAM on a more permissive license and letting other projects share the code (and avoiding this whole GPL incompatibility headache), but somebody would have to track all the contributors and ask them what they think about is. If someone wishes to keep the project on GPL license, I completely understand that too.

pppalain, 2021-05-01:
this is what Vilem Novak wrote:
# This program is free software; you can redistribute it and/or
# modify it under the terms of the GNU General Public License
# as published by the Free Software Foundation; either version 2
# of the License, or (at your option) any later version.

so everything can be changed to GPL3 at teh discression of the one using it

closed 2023-01-14 by pppalain