vilemduha/blendercam #160: `Read-only file system` error on MacOS

illus0r, 2020-09-09:
Hello there! Thanks for the free and powerful tool!

It looks fascinating, but when I try to calculate path, I get the error:

[image]

Both Blender and its python are in the list of full disk access apps:

[image]
[image]

How is this solvable?

ksuprynowicz, 2020-09-09:
Hi!
Have you saved the Blender file to some directory where you have write access before generating gcode?

illus0r, 2020-09-09:
@ksuprynowicz thank you! It did the trick!

closed 2020-09-09 by illus0r

ksuprynowicz, 2020-09-09:
Nice to hear that :)