vilemduha/blendercam #161: I am failing to install 2.80 blender cam on windows. and 5-axis cnc .

mir132, 2020-09-25:
I am failing to install 2.80 blender cam on windows.
I would like to know how to install the latest version of Blender cam on Windows.
I would appreciate it if you could explain it in detail as you would explain it to a child.
And does Blendercam currently support 5-axis CNC?

TurBoss, 2020-09-25:
Hello,
there is no guide to install opencamlib on windows which is a blendercam requirement

you can try to adapt the linux steps to run on windows maybe using msys2? mingw?

this are the linux steps
https://github.com/vilemduha/blendercam/wiki/Using-BlenderCAM-with-OpenCAMLib
https://github.com/vilemduha/blendercam/wiki/Install-opencamlib-into-blender-2.80-on-linux

PD:
4 and 5 axis are experimental features and may not work at all
I tried 4 axis without great success

pppalain, 2021-01-12:
there is another option: install a virtual machine and install linux....  Opencamlib is now required if you would like to use the ballcone cutter

pppalain, 2021-01-12:
regarding 5 axis I have made some modifications to be able to work with a A and B axis.  these are not automatic but you may rotate your model and then make your strategies.... so rotate A 45 degrees, use parallel strategy, rotate 90 use parallel strategy.  so it's a compromise, far from perfect usually good for chaining operations.

pppalain, 2023-01-14:
new version of blendercam autoinstall dependencies

closed 2023-01-14 by pppalain