vilemduha/blendercam #165: Does the program have one-touch drawing function?

mir132, 2020-11-08:
Does the program have one-touch drawing function?

ksuprynowicz, 2020-11-08:
What do you mean by one-touch drawing?

mir132, 2020-11-09:
> What do you mean by one-touch drawing?
@ksuprynowicz
For example, it is also called the Eulerian trail.
I want to build a toolpath that follows a curve.
When using a profile on an inclined sample plane, only the outline is created on the floor.
I want to get a skewed sample or a toolpath that follows a curve.

ksuprynowicz, 2020-11-09:
Sure! I needed it a few times and it works very well, also for 3D curves. You have to enable experimental features in plugin option and it's something like "curve to path". As you said, "profile" wont work as intended. Let me know if you can find it, if not I'll check how it is called in the evening.

mir132, 2020-11-09:
@ksuprynowicz 
Can you see where the plug-in is located by picture? "curve to path" 
Thanks for your help. 
I am using version 2.76.
I want to install the latest version in a window environment.
It fails every time.

pppalain, 2023-01-14:
so version 2.76 is not supported.  easier to install on windows with new version.  dependencies autoinstall.

closed 2023-01-14 by pppalain