vilemduha/blendercam #166: RuntimeError: Operator bpy.ops.object.select_all.poll() failed, context is incorrect

Pingear, 2020-11-15:
Hi)
When i click "Calculate path", throw this exception -

Traceback (most recent call last):
  File "/home/user/Desktop/Blender/blendercam-2.80_BETA/scripts/addons/cam/ops.py", line 213, in execute
    bpy.ops.object.select_all(action='DESELECT')
  File "/home/user/Desktop/Blender//blender-2.83.9-linux64/2.83/scripts/modules/bpy/ops.py", line 201, in __call__
    ret = op_call(self.idname_py(), None, kw)
RuntimeError: Operator bpy.ops.object.select_all.poll() failed, context is incorrect

location: <unknown location>:-1

Сan anyone explain what the problem is?

mMcFab, 2020-12-10:
Hello!
In personal testing, I have found that "Calculate Path" currently only works in Object mode, otherwise this exact message is shown (with different file paths, at least) - could this be the issue for you as well? 

I hope this helps!

ksuprynowicz, 2020-12-11:
Hello, and sorry for replying so late,
The calculate path should be used only in Object mode indeed. If it still doesn't work, you can provide a sample file and I will take a look at it.

Pingear, 2020-12-11:
Unfortunately I deleted problem project, and i probably can't reproduce this error. So i close this issue. Thanks for reply)

mMcFab, 2020-12-11:
Ok, so I "fixed" this in my fork, but I'm not sure if I should make a PR for it and I'm not even certain I _can_ with a free/personal account because I made a lot of UI changes before this and I can't make a "bugfix" fork (although this is pretty much my first proper experience with git/github so I'm not sure how most of it works yet, maybe I can make something that can be integrated? Maybe a bugfix-only branch on my fork? I did do a couple of other fixes that could be useful, like chain simulation not detecting paths correctly)

My fix was to just hide the calculate buttons (and tell the user why) in the `ui.py` with code similar to this:
```
if(bpy.context.mode == 'OBJECT'):
	# Draw the operator button
else:
	layout.label(text="Calculate is only in Object Mode")
```
Which seems to work well enough and should prevent this error coming up again unless it's under a different set of circumstances and is therefore a different bug
An alternative could be to automatically switch to object mode for the calculation, then back to the previous mode, but I don't know if there are certain contexts where this would upset Blender, so that might be asking for trouble. 

I hope this is useful and that I'm not posting it in the wrong sort of place, I'm new to this

ksuprynowicz, 2020-12-15:
Thank you! I will take a look at it tomorrow.
What UI changes have you made? Maybe it would be a good idea to merge some of them too.

pppalain, 2021-04-07:
instead of graying out, I changed ops to force object mode during calculate path
The next pull will include that.
bpy.ops.object.mode_set(mode = 'OBJECT')	#force object mode

pppalain, 2021-04-07:
issue now fixed

closed 2023-01-14 by pppalain