vilemduha/blendercam #168: [improvement proposal] add G64 gcode path blending option with tolerance in add-on menu

Olm-e, 2020-12-02:
following the #83 issue and it's proposed solution, add a checkbox and a field in the add-on menu to validate the use and set the tolerance of the G64 code for path blending.
it looks like a needed option for most cuts as arcs are not supported, so to "not burn bits at first cut 'cos you didn't knew about it" ... 
I have from my side validated that just adding the G64 P0.015 parameter on a regular precise enough 18mm / 6mm bit wood cut _helps a lot_ compared to a raw gcode output without path blending (P0.015 could probably be default value)

pppalain, 2020-12-19:
current pull request addresses this issue.  will be added to CAM movement menu

pppalain, 2020-12-19:
[image]

pppalain, 2021-01-12:
today's merge has fixed this issue
[image]

closed 2023-01-14 by pppalain