vilemduha/blendercam #169: I wish to develop blender cam

abosafia, 2020-12-16:
I wish to develop blender cam and don't know where can I start
First I want to add post processor
Second I want to add v_carve operation
Thanks

TurBoss, 2020-12-16:
hello and welcome 

> I wish to develop blender cam and don't know where can I start

you could make a fork of the repository in your account, then create a branch named "postprocesor_for_x_machine",
then push commits to that branch, then you can open a pull request from your fork branch to this repository

> First I want to add a post processor

postprocesors can be added to scripts/addons/cam/nc
there are many that can be used as example

>  Second I want to add v_carve operation

I think that you can save new operation presets

ksuprynowicz, 2020-12-16:
Let us know if you get stuck on something.
I'll be happy to help.

abosafia, 2020-12-16:
Thanks a lot 
My problem with blender cam happens when I use profile operation (outside) for shapes with right angle 
Let's say for a rectangle 
The path at this corner is rounded 
I'm using mach3
When I cut this piece it comes out with rounded corners
So to prevent this I have to get an offset of my shapes by cam tools and edit those rounded corners in edit mode then do profile cutout (on line) and this solve the problem
In other programs like artcam or aspire
The problem can be solved by choosing special post processor
So I thought that if I could add those post processor I can get red off that problem
Sorry for too much talking 
Thanks a lot for offering help

abosafia, 2020-12-18:
> postprocesors can be added to scripts/addons/cam/nc
> there are many that can be used as example
Can you please show me how to get a post processor and add it in steps
The post processor I want to add  is mach3 arc
In artcam

TurBoss, 2020-12-18:
https://github.com/vilemduha/blendercam/blob/master/scripts/addons/cam/nc/mach3.py
https://github.com/vilemduha/blendercam/blob/master/scripts/addons/cam/nc/iso.py
https://github.com/vilemduha/blendercam/blob/master/scripts/addons/cam/nc/nc.py

this is the match3 post we have and it just uses ISO and simply adds tool compensation to the tool changes
if you add new custom match3 needs there

abosafia, 2020-12-18:
Thanks
I will try and see if this solve my problem

abosafia, 2020-12-18:
[image]
This is the problem I wish to get red off
when cut this with mach3 it comes rounded 
higher feed rate makes the rounded corner radius longer
My solution is to edit this path like the next image
[image]
or by using cam tools to get offset curve with (my tool radius) then edit the rounded corners 
then I use profile cut online
and the gob is( done)

abosafia, 2020-12-18:
The same problem happens with artcam but when I chose mach3 arc post processor the  outside cut path came with perfect right angle

abosafia, 2020-12-18:
> https://github.com/vilemduha/blendercam/blob/master/scripts/addons/cam/nc/mach3.py
> https://github.com/vilemduha/blendercam/blob/master/scripts/addons/cam/nc/iso.py
> https://github.com/vilemduha/blendercam/blob/master/scripts/addons/cam/nc/nc.py
> 
> this is the match3 post we have and it just uses ISO and simply adds tool compensation to the tool changes
> you can add new custom match3 needs there on the match3.py file

I tried those post processor and nothing happened

pppalain, 2020-12-19:
for v carve use experimental medial axis.  I have not tried it

abosafia, 2020-12-19:
I will
Thanks

bostjannagode, 2020-12-19:
Anyone thinks for this (using Linuxcnc)

G21 = mm
so G64 P0.001 = 1 micron deviation max (I wonder if this triggers a bug)

What happens with G64 P0.01 ?

bostjannagode, 2020-12-19:
I write in manual command 
P 1.0 meand trayectory +-0.5mm from planned. On corners, its much faster as it doesnt neex axis to wait for smooth stop...

pppalain, 2020-12-19:
I have a pull request modifying the G64 right now.  you can check it out.  it will be set in the CAM motion panel

pppalain, 2020-12-19:
[image]

abosafia, 2020-12-20:
> for v carve use experimental medial axis. I have not tried it

It doesn't work
I using a program called f-engrave to do v carving operations 
if you can put this operation in blender cam would b magnificent

abosafia, 2021-06-08:
My wish came true. Thanks to Alain (pppalain) for helping. Now there is an option to choose join style to get squared corners not rounded. And medial axis ( vcarve) began to work. Join us https://riot.im/app/#/room/#blendercam:matrix.org

abosafia, 2021-06-08:
Done

closed 2021-06-08 by abosafia