vilemduha/blendercam #170: Issue with imported dxf file

Franckapik, 2020-12-17:
Hi! I would like to understand why i have an error on cutout blendercam. 
The shape of my imported dxf seem's to be wrong ... :/
https://pasteboard.co/JFp2hlF.png
and this is the error : https://pasteboard.co/JFp2Mjr.png
I have clean up with remove double and merge by distance
but it seem's to have another issue...
Is there a command to be sure that every vertex are merged ? Someone had an idea of what is causing this issue ?

Thank you !

Franckapik, 2020-12-17:
Here is my blend file : https://pasteall.org/blend/9f65fbeda3ca479a80f8ea2655217e90

TurBoss, 2020-12-17:
hello,
there is a unmerged vertex at the bottom left
i and only after creating faces for each part it allowed me to calculate

[image]
[image]

[pirate.zip](https://github.com/vilemduha/blendercam/files/5712616/pirate.zip)

closed 2021-04-29 by Franckapik