vilemduha/blendercam #174: Installed BlenderCam into 2.82a but can't figure out how to make it work

ctzsnooze, 2021-02-03:
Hi
Sorry to be annoying.
I have successfully installed the current Github version of BlenderCam on MacOSX, but I can't seem to figure out how to make it work.  Everything is in different places in 2.8 compared to 2.7.
I seem to be able to configure the machine preset and can visualise the CAM machine in my Layout.
But path generation fails even with a simple object.
Is there a usage guide for 2.8 that shows exactly what to do?  There are excellent notes for 2.7x builds of Blender, but I can't find anything for 2.8 or higher.
Any advice greatly appreciated.
Chris

ctzsnooze, 2021-02-03:
Getting somewhere - in Mac OSX, the scripts need to be located inside the Blender 2.8 application contents, and can't be externally referenced, otherwise there are operation not permitted errors.

pppalain, 2021-03-09:
can you install linux in a virtual machine?

closed 2023-01-14 by pppalain