vilemduha/blendercam #180: Problems with the new API

viralatadesbois, 2021-03-19:
Hi, i'm managing to get most of blendercam working with 2.92, but I still have some issues:
Silhoute is not working.
 It is easily fixed by replacing cursor_location by cursor.location in ops.py

But slice.py has another another error that I'm not sure I can fix:
by replacing depsgraph by evaluated_depsgraph_get it fixes part of it but then I get "TypeError: Object.to_mesh(): takes at most 2 arguments, got 3" and I'm not confident enough to go further.

If anyone knows how to fix this...

Thankyou

viralatadesbois, 2021-03-19:
I forgot to mention that the bridges are not working anymore too, because the groups have been replaced by collections so we get the error: AttributeError: 'BlendData' object has no attribute 'groups'

richieloro, 2021-03-30:
What version of blender do you recommend for the current Blendercam master?
I can't install blendercam in blender, when I activate the addon by checking the box, blender crash.

I downloaded blender-2.92.0-linux64.tar.xz from the blender website.
I installed shapely in blender with the built in python3.7m :
```
cd ~/path/to/blender/2.92/python/bin
./python3.7m -m ensurepip
./python3.7m -m pip install shapely
```

I added the blendercam script folder path in blender file preferences, saved.
Restarted blender.
But when I activate the blendercam addon, blender 2.92 crashes, and close.
Merci.

pppalain, 2021-04-04:
let me check join us on The Matrix (#blendercam:matrix.org) for chatting, this is faster as I do not check this forum all the time.

pppalain, 2021-04-04:
I confirmed that bridges work and that silhouette work will try slice.py as I am not familiar with it.

pppalain, 2021-04-04:
I tried slice.py and got into a bad loop generating endless slices and text for those slices. I did not get your errors. but still was unable to make it work.  Please note that the latest version is in the pull request right now with several commits.

pppalain, 2021-04-04:
It appears you are trying to get the old 2.76 version of blendercam wroking on 2.92.... you are missing 2 years of updates.  but still slice.py makes a loop error with the old code.

pppalain, 2021-04-04:
working on rewriting slice.py with new functions.  It was originally written in 2012 so new blender features are pertinent now.... will be faster and hopefully as good.

pppalain, 2021-04-05:
slice.py was completely rewritten and is available on the pull request

closed 2023-01-14 by pppalain