vilemduha/blendercam #181: Ubuntu 20.10 - Blender 2.92 crashes when I try to activate Blendercam from github

richieloro, 2021-03-30:
What version of blender do you recommend for the current Blendercam master?
I can't install blendercam in blender, when I activate the addon by checking the box, blender crash.

I am on Ubuntu 20.10.
I downloaded blender-2.92.0-linux64.tar.xz from the blender website.

I installed the dependencies with `apt install python3-shapely python3-numpy`

I installed shapely in blender with the built in python3.7m :
```
cd ~/path/to/blender/2.92/python/bin
./python3.7m -m ensurepip
./python3.7m -m pip install shapely
```

I added the blendercam script folder path in blender file preferences, saved.
Restarted blender.
But when I activate the blendercam addon, blender 2.92 crashes, and close.

Note the same thing happens when I manually copy the content of the script in the blender addon folder. It crashes when I tick the box.

Please help me. Thank you

ad1217, 2021-03-31:
In a fresh Ubuntu 20.10 VM, Blender crashes when attempting to enable blendercam with `Aborted (core dumped)`, but no other helpful information (even with `--debug-all`). When running `./blender-softwaregl`, however, enabling works fine. If blendercam is enabled in software GL mode, normal `./blender` will crash on startup.

Possibly related: https://developer.blender.org/T81720, which further suggests that it is a graphics issue and somehow specific to Ubuntu 20.10 (but probably not actually a blendercam bug).

richieloro, 2021-03-31:
Thank you so much ad1217 you don't know what it means to me to be helped like this.
Thank you very much. 
So it is an ubuntu bug it seems, I also have intel gpu like the bug report author.

I just left windows after 10years and started using linux seriously and for personal and work.
I made a lot of efforts, compromises and learned ubuntu seriously, I almost went back to windows multiple times in the last months.
But seeing how the open source community work and think, the help I received from you and the others. I really feel safe by completely going opensource. I may settle on Arch linux because your build of blendercam works on Manjaro.

Thank you guys for the efforts you put in your work, and the love you share it's heartwarming.

pppalain, 2021-04-04:
I finally left windows completely in October after their update bootlooped my PC.  If you have any questions we have an active group on The Matrix (#blendercam:matrix.org) and we are continuously trying to improve blendercam

richieloro, 2021-04-11:
Thank you I will join the group.  I just installed opencamlib on my system. 
I think blendercam lacks tutorials online, I will try to make some if I master it.

closed 2023-01-14 by pppalain