vilemduha/blendercam #186: Install on Windows

s08glen, 2021-05-24:
Hi,
I am a student and I want to use Blendercam to make some animation for my project.
I want to install this cam on Windows because some of my side project is depend on Windows OS.

I knew that there are some Python package need to be installed before using it, so I followed the following unoffical youtube video and failed.
https://www.youtube.com/watch?v=IDS8GNq1LTE
https://www.youtube.com/watch?v=k4y3rRAvfl0

When I tried to enable the "Scene:CAM" in Blender Preferences, it show me the error message such like "ImportError: DLL load failed: Cannot find module" " Traceback -> import numpy -> from . import core -> raisw ImportError(msg)" " Import the numpy C-extensions failed", I also cannot turn CAM Render Engine in Blender Scene.
I have checked numpy and shapely had been installed correctly.

Is there any other way to install this project in Windows environment instead of Linux OS? 
Thanks a lots!

pppalain, 2021-05-29:
join https://riot.im/app/#/room/#blendercam:matrix.org chat and usually someone can help you there in various time zones.

hroma, 2021-09-06:
I've also failed to install blendercam under windows.
Suggestion: 
Blender as a portable version for windows https://www.blender.org/download/Blender2.93/blender-2.93.4-windows-x64.zip/ . Wouldn't be possible that someone (else than nooby like me) manage to include all dependencies required to a portable version? Windows users would be so gratefull :-)

abosafia, 2021-09-06:
join https://riot.im/app/#/room/#blendercam:matrix.org chat and usually someone can help you there in various time zones.

wohali, 2022-06-16:
In the attached zip is the `ocl.pyd` file you need for the current Blender 3.2, Windows 64-bit, embedded Python 3.10.2. 

Put this in the `Blender 3.2\3.2\python\lib\site-packages` directory of your installation.

Hope this helps someone - took me a day to sort out compiling it....

[ocl.zip](https://github.com/vilemduha/blendercam/files/8923087/ocl.zip)

pppalain, 2022-06-19:
I recently installed on version Blender 3.1 on windows without OCL and posted a video. https://youtu.be/RXpGLJz7ZIs OCL is optional and new versions of blender have better collision making OCL really optional.

wohali, 2022-06-20:
> I recently installed on version Blender 3.1 on windows without OCL and posted a video. 

Fortunately this part is easy, or at least I think so :) thank you for sharing this for others!

I wanted to mess with waterline and OCL.

abosafia, 2022-08-17:
> 

Water line now works with ocl
You can try
https://github.com/pppalain/blendercam

abosafia, 2022-09-19:
> In the attached zip is the `ocl.pyd` file you need for the current Blender 3.2, Windows 64-bit, embedded Python 3.10.2.
>
> Put this in the `Blender 3.2\3.2\python\lib\site-packages` directory of your [...]

Thanks for the file. It's working very good. But dependencies have to be installed.

joemarshall, 2022-12-28:
https://pypi.org/project/opencamlib/

In case anyone is looking for it, it's on pypi as an installable package now.

joemarshall, 2022-12-28:
https://pypi.org/project/opencamlib/

In case anyone is looking for opencamlib  it's on pypi as an installable package now. 

Blendercam on pypi seems like an obvious next step now

joemarshall, 2023-01-11:
Install from https://github.com/pppalain/blendercam

and it works fine on windows

pppalain, 2023-01-14:
new version of blendercam now autoinstalls dependencies

closed 2023-01-14 by pppalain