vilemduha/blendercam #189: Selecting Ball presets doesn't update the tool settings and throws an error.

ecv80, 2021-06-28:
Blender 2.93.1

Selecting Ball presets doesn't update the tool settings and throws the following:

Error: Failed to execute the preset: TypeError('bpy_struct: item.attr = val: enum "BALL" not found in (\'END\', \'BALLNOSE\', \'BULLNOSE\', \'VCARVE\', \'BALLCONE\', \'LASER\', \'PLASMA\', \'CUSTOM\')')

It looks like it's simply a matter of renaming some entries.

ksuprynowicz, 2021-06-28:
Hi!
I just accepted two pull requests with various fixes. Can you try again and let me know if it works now?

ecv80, 2021-06-28:
Oh, unfortunately it doesn't appear to work, but maybe I didn't update properly. This is all I did: "git pull" and I restarted Blender. Is that enough? Thanks.

ecv80, 2021-06-28:
Also is there a place to have a brief chat? I'm pretty much new to CNC milling and I have a question regarding roughing and finish operations in BlenderCAM.

pppalain, 2021-06-30:
that was because ball was deleted and not supported by opencamlib.  use ballnose instead

closed 2023-01-14 by pppalain