vilemduha/blendercam #191: Blender 2.93 crashes when exabling OpenCamLib

dexington, 2021-07-17:
I've installed BlenderCam and OpenCamLib according to the latest instructions under Ubuntu 20.

I'm able to generate a parallel toolpath and simulate it with 'exact mode' enabled.
However, when enabling OpenCamLib, blender crashes. Strangely, it doesn't generate a blender.crash.txt file. 

I noticed one error when installing OpenCamLib using "cmake -DBUILD_PY_LIB=ON -DBUILD_CXX_LIB=ON -DUSE_PY_3=ON .."
as follows:
"Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)"
Other than that I didn't see any errors.

abosafia, 2021-07-17:
hi dexington
there is somthing wrong with your model I don't know 
I exported it as stl 
then imported it  in new file works fine
join us in
https://riot.im/app/#/room/#blendercam:matrix.org
we can help better
[KuduQ Engrave test .blend.tar.gz](https://github.com/vilemduha/blendercam/files/6834886/KuduQ.Engrave.test.blend.tar.gz)

dexington, 2021-07-17:
@abosafia Thank you. I will join you there.

TurBoss, 2021-07-17:
hello

"Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)"

```
sudo apt install doxygen
or doxygen-dev
```
but this is optional to generate docs

dexington, 2021-07-17:
@TurBoss Thank you.

closed 2023-01-14 by pppalain