vilemduha/blendercam #192: Simulation failed, if combination of file name and cam operation name is to long

aak67, 2021-07-24:
When you start a simulation, an .exr image will be generated in the temp_cam folder. The .exr file name is a combination of
Blender filename + "_cam_path_" + cam operation name + "_sim.exr"
Blender allow in the function bpy.ops.image.new() only 63 characters
If Blender find object with an existing name the old object name will be modified with a trailing number. e.g. .002
Because of this your combination of file name and cam operation name should not be longer than:
`63 - ".002" - "_sim.exr" - "_cam_path_" = 63 - 4 - 8 - 9 = 42
`
Otherwise you will get an error like this in the console:

```
 Python: Traceback (most recent call last):
  File "/usr/share/blender/2.93/scripts/addons/cam/ops.py", line 390, in execute
    simulation.doSimulation(operation_name, [operation])
  File "/usr/share/blender/2.93/scripts/addons/cam/simulation.py", line 108, in doSimulation
    image_utils.numpysave(i, iname)
  File "/usr/share/blender/2.93/scripts/addons/cam/image_utils.py", line 79, in numpysave
    i = numpytoimage(a, inamebase)
  File "/usr/share/blender/2.93/scripts/addons/cam/image_utils.py", line 115, in numpytoimage
    i.pixels[:] = a[:]  # this gives big speedup!
UnboundLocalError: local variable 'i' referenced before assignment
```

In image_utils.py the function numpytoimage(a, iname) try to find the an already rendered image and create a new image if not found. But it will never find the image, because iname contain the full unique name and blender knows only the shortened one. The bpy.ops.image.new() function cut everything longer than 63 characters. 

Solution
I have no good idea. May be creating a mapping table and use uuid:
[docs.python.org UUID](https://docs.python.org/3/library/uuid.html)
example:
```
>>> str(uuid.uuid4())
'e859d72f-2437-456c-9267-81c9f92a538e'
```

abosafia, 2021-07-29:
Hi aak67
Join us in 
https://matrix.to/#/#blendercam:matrix.org

pppalain, 2021-09-09:
issue fixed in pull #194 August fix and improvements

closed 2023-01-14 by pppalain