vilemduha/blendercam #195: Instalation on Snap version of Blender

2Eagle2, 2021-09-18:
Hi
Firstly thanks for great addon !!!

I have some issue when installing on snap version of blender
it has its own embedded python which is on read only filesystem (snap) so required missing libraries cannot be installed directly there.

I can use 
```
./python3.9 -m ensurepip --upgrade
./python3.9 -m pip install shapely equation
```
and it install them into user directory /home/${USER}/.local/lib/python3.9/site-packages

the problem is that the blender do not see that packages anyway

Is there recomended way how to install this plugin to snapped blender

2Eagle2, 2021-09-18:
I have created small hack to make it work !!!
I have created file preinit.py in blendercam/scripts/addons/cam/
```
import sys
import getpass

pythonver=f"python{sys.version_info.major}.{sys.version_info.minor}"

user_py_path = f"/home/{getpass.getuser()}/.local/lib/{pythonver}/site-packages"
if user_py_path not in sys.path:
    sys.path.append(user_py_path)
```
and then in blendercam/scripts/addons/cam/__init__.py on the beginning I have added
`from . import preinit`

that make the plugin work, but I consider it as a hack, but at least it works

jnewbigin, 2022-03-17:
I managed to work around this same problem by creating symlinks in the `scripts/addons/` directory
```
ls -l scripts/addons/
Equation -> /home/john.newbigin/.local/lib/python3.10/site-packages/Equation
shapely -> /home/john.newbigin/.local/lib/python3.10/site-packages/shapely
```

davisburnside, 2022-05-27:
You can tell pip to install to a specific directory. 
https://intellipaat.com/community/9428/install-a-python-package-into-a-different-directory-using-pip

The folder you want to install to is the "site-packages" folder inside the Python distro that comes with blender, Here's how to find it
https://b3d.interplanety.org/en/path-to-blender-python-executable/

If there is a mismatch between your computer's Python and Blender's Python version, you may need to use Blender's Python executable to execute the pip command.

closed 2023-01-14 by pppalain