vilemduha/blendercam #198: 'Opencamlib NOT installed'

d8sconz, 2021-11-30:
I can't install Opencamlib for some reason. It is installed, simlinks in place, alternative script path in preferences. I've checked, double checked and repeated the whole setup process but I continue to get this message under 'CAM info & warnings' tab. I'm using Blender 2.93.1 on Linux Mint 20.2.

neuroblossom, 2022-04-17:
Im having the same issue, 'Opencamlib NOT installed'  Blender 3.1.0 MacOS Monterey 12.3. Have also tried copying the .py and .so library files into my blender python site-packages folder as well as symlinking.

abosafia, 2022-04-17:
I'm using Linux mint and it is installed.
Join us at 
https://matrix.to/#/#blendercam:matrix.org

pppalain, 2023-01-14:
new version auto installs dependencies including shapely

closed 2023-01-14 by pppalain